raulmur/ORB_SLAM2 #235: vocabulary.txt is it  made using all point descriptor?

chaiein, 2017-01-17:
How to construct the vocabulary.txt file? will it change if the line descriptors are used instead?

AlejandroSilvestri, 2017-01-20:
I don't know what is a line descriptor.

vocabulary.txt contains a tree for ORB descriptors classification into words, bag of words (BoW).  ORB-SLAM2 uses it to form a Database of BoW (DBow).  It was generated by a wibbly wobbly timey wimey machine learning stuff.

If you want to build your own vocabulary, I believe you should start from [this paper](http://doriangalvez.com/papers/GalvezTRO12.pdf).

chaiein, 2017-02-21:
Vocabulary.txt is constructed using FAST descriptors which are not line descriptors. Since in this ORB_SLAM , ORB features are extracted we can use the vocabulary.txt provided in this opensource. If I also extract the features using EDL that are lines from the image, then during matching vocabulary.txt provided in the opensource is not useful. I should construct a new vocabulary.txt . Am I right ??

AlejandroSilvestri, 2017-02-22:
You should contruct a new vocabulary.txt and a new SLAM to use it.

[EDL](http://docs.opencv.org/3.0-beta/modules/line_descriptor/doc/line_descriptor.html) [paper](http://www.ipol.im/pub/art/2012/gjmr-lsd/article.pdf)

ORB-SLAM2 detects features with FAST and extracts descriptors with a particular BRIEF implementation from ORB.  BRIEF are binary descriptors, in this case 256 bits binary descriptors, and their distance is evaluated as Hamming distance with Stanford's algorithm coded for that exact length: 256 bits.

On top of that ORB-SLAM2 uses a vocabulary tree of these kind of descriptors, and crawls it repeatedly evaluating Hamming distance with aforementioned algorithm.

I believe you'll also use a 256 bits binary descriptors so you can reuse some of the code.  But geometry is completely different, it will be a new SLAM, with new code and deserving a new paper.

Fatmoham, 2017-07-19:
@raulmur , @AlejandroSilvestri 
As mentioned in this paper "Fast Relocalisation and Loop Closing in Keyframe-Based SLAM" you created vocabulary using the data-set Bovisa 2008-09-01. This Vocabulary is taken from a huge sets of data and works pretty good for indoor and outdoor SLAM.
If I want to use the same program ORBSLAM2 in completely different environment rather than indoor and outdoor environment, are the ORBvoc.text still good to use? Or I need to generate my own Vocabulary for the data-sets I collected from that environment?

AlejandroSilvestri, 2017-07-20:
@Fatmoham 

I believe provided vocabulary will work on any environment.  You don't need to generate your own vocabulary to get started.

With a better understanding of BoW and knowledge about your specific environment, you could discover that creating a better and more suitable voc can improve feature matching.

Fatmoham, 2017-07-20:
@AlejandroSilvestri 

The problem about my data-sets is that for each run of ORBSLAM2, I got different camera trajectories  (not consistent trajectories). any suggestion to improve my results?

AlejandroSilvestri, 2017-07-20:
@Fatmoham 

Can you publish a video of this dataset?  Perhaps someone with experience can give you a tip.

In my experience, not consistent trajectories are usually related to few or poorly distributed map points (wrong vocabulary is a possible cause, but not the only one), but  are more often caused by bad camera calibration.

Fatmoham, 2017-07-20:
@AlejandroSilvestri 
what about the features that was extracted from moving (dynamic)object in the scene.  if the majority of features were extracted from moving object, how ORBSLAM2 will be able to estimate the accurate camera trajectory?

AlejandroSilvestri, 2017-07-20:
In general, SLAM is meant to work on static scenes.

Many SLAM solutions are robust, and can discard moving objects features, provided they are not dominant in the scene.

Moving objects can alter your trajectory.

ArtlyStyles, 2018-09-30:
what is the size (how many words) of the ORB-SLAM vocabulary ?

AlejandroSilvestri, 2018-10-05:
@ArtlyStyles, well, not exactly.

The match in 1e6 word vocabulary is only the first step, before further matching.  There are 7e16 different descriptors, you don't look for an exact match.  Bag of words is used to narrow down brute force descriptor matching.

In BoW you aren't looking for the best, you are looking for an exact match.

In bag of words, words are integers (20 btis each in this case) that match exactly, with equal value.  This is different to descriptor, where you must compute distance and accept low distance as a match.  When you match equal values from a list, you order the list and perform at least a binary look up, starting from the middle, asking if it is lower or greater...  20 comparisons for 20 bits.  This way it actually is very affordable.  And it improves when you are looking for a bag of words, not only one word.

craftindo, 2020-11-13:
I'm still confused with this bag of words algorithm.
1. if it is looking for an exact match, then is the vocabulary growing each time I feed a new scenery?
2. If yes, the processing speed will grow slower as it processes new images.
3. the paper said it is built offline. So, number 1 doesnt make sense at all.
4. what make sense is that, bag of words algorithm is matching a scene to a class not a perfect match descriptor matching the scene. i guess.

thanks in advance for answering my questions

AlejandroSilvestri, 2020-11-13:
@craftindo 

Number 4.

Vocabulary is a mapping from descriptor to **word**.  Many descriptors with low distance map to the same word.  The grouping was done offline with a decision tree classifier.  So, on a huge dataset with descriptor already matched, these fellows trained a machine to classify descriptors into a number of words, each word representing a class in machine learning jargon.

Then you describe a scene (a frame) as a set of words (a bag of words).  So, if you are looking for a similar scene -for relocalization or loop closure purposes- you compare their words and count how many word matches they have.  This comparison is very fast.  Keep in mind in this cases you aren't looking for keypoint matches.

If you already have the words computed, brute force matching is faster on words than on descriptors.  When looking for matches you can narrow the search:

1- epipolar constraint: only consider keypoints near an epipolar line
2- words: only consider candidates to match when both keypoints have the same word
3- compute distance: get the best distance provided an absolute maximum threshold and a relative minimum threshold to the second best distance

TonyPhh, 2021-12-10:
> @Fatmoham
>
> I believe provided vocabulary will work on any environment. You don't need to generate your own vocabulary to get started.
>
> With a better understanding of BoW and knowledge about your [...]

hello,do you konow how to get the 'Bovisa 2008-09-01' dataset, I find I cannot download the dataset from  the official website. thanks!

supermice, 2021-12-10:
邮件已收到！ ***@***.***

fightmoney, 2022-09-30:
@TonyPhh 你能下载这个数据集了吗? 我在官网也下不了，请问怎么下

fightmoney, 2022-09-30:
@supermice 你能下载这个数据集了吗? 我在官网也下不了，请问怎么下