raulmur/ORB_SLAM2 #237: Has anyone had successfully Pangolin working on iOS ?

hoverpeter, 2017-01-20:
I made a simple iOS App that builds on the core engine, but still missing graphics module.

pmoulon, 2017-01-20:
Nice,
Not sure that you can publish your code under MIT license since ORB_SLAM is GPL...

hoverpeter, 2017-01-20:
@pmoulon thanks. I apologize. I'll make changes upon your comment. Any guidance?

pmoulon, 2017-01-20:
No worries.
You must use the same license as the original project
You must include this file https://github.com/raulmur/ORB_SLAM2/blob/master/License-gpl.txt
And you can add another file explaining the goal of your project as it is done here https://github.com/raulmur/ORB_SLAM2/blob/master/LICENSE.txt

FYI.
https://choosealicense.com/licenses/ 
GNU GPLv3 => Condition => Same License

closed 2017-01-20 by hoverpeter