raulmur/ORB_SLAM2 #238: System crashes when execute ./build.sh

Kanelankai, 2017-01-23:
I use VirtualBox in my Mac to build a ubuntu 14.04 and have installed all the required dependencies as section 2 says. However, when I execute ./build.sh of ORB_SLAM2, PC crashes everytime with a scene as follows:
[image]
How can I solve this problem?
Thanks,please help @raulmur

Kanelankai, 2017-01-23:
[image]

tugyan, 2017-01-23:
I do not know if your case is same with mine. But in my case, compilation crashes with out of memory error. I think this is due to some bug in compiler when compiling with multiple cores. To disable parallel compiling, you can do following changes in build.sh file: Remove every "-j" argument that comes after "make" command so that every "make -j" line will be only "make" without quotes. Hope this helps.

nmerrill67, 2017-01-23:
The -j option tells the compiler how many cores it can use. If you have four cores (or an i7 with two) I would suggest make -j4 for every make command. I have a similar setup, and I have compiled it many times now.

Kanelankai, 2017-01-24:
Thank you for your kind reply. @tugyan 
I removed every "-j" as you said and executed the ./build.sh file again and it works!
Now I think I can do some tests using tum or kitti.
Thank you.

Kanelankai, 2017-01-24:
Thank you very much! @nmerrill67 
I removed every "-j" in the ./build.sh file and it works!

iDonghq, 2017-07-27:
Have you run this problem which when I test “./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml PATH_TO_SEQUENCE_FOLDER ASSOCIATIONS_FILE ” ,and then nothing happened. Could you help me solve this problem?

Kanelankai, 2017-07-27:
Have you changed "

PATH_TO_SEQUENCE_FOLDER ASSOCIATIONS_FILE"

to the path of the file on your PC?