raulmur/ORB_SLAM2 #239: Failing to initialize in low fps planar scenes

laxnpander, 2017-01-24:
Hey there,

i am currently trying to use the ORB SLAM in UAV applications, especially outdoor. So flying at 100 m heights and tracking the position visually. Therefore planar scene with bad depth informations. My camera has about 1-2 fps but with high overlap between consecutive frames. In this scenario ORB SLAM fails to initalize.
I would like to understand what the main cause of this might be and what possibilities I have got? Has anyone experiences to share?

Best regards

SamuelDudley, 2017-01-24:
Hi, 
I'm also using ORB-SLAM2 for fixed wing UAS applications over long distances (~100km return). Altitudes of my data sets are ~500m AGL and camera frame rate is ~5Hz. Once the aircraft is in the air I can initalize  while doing a slow loiter circuit, however I have also found that during the long straight segments (cruise) the system will struggle / fail to initalize all together. I have uploaded a [short video](https://youtu.be/sgYyfJT_oK8) (skip to towards the end) where you can see the system initializing during a loiter.

Regards,
Sam

laxnpander, 2017-01-25:
Very nice and interesting stuff! One thing I see is, that you extract way more features than I do / find more matches than I do. I will check if its a software (settings/extraction/matching) or camera problem. How much afford do you spend on calibrating the camera? 
As soon as I am back home I will upload a video of my data set. In my case (2 axis gimbal stabilized images) it could be an option to initialize with rigid transformation (only allowed x,y, rotation around 1 axis and scaling) instead of the full 8 dof of homography, what do you think?

SamuelDudley, 2017-01-26:
I'm using `ORBextractor.nFeatures: 3000` in the .yaml launch file which might help. Depending on your target hardware you might be able to get away with more as your frame rate is quite low (like mine).
I printed up a very large calibration target and used ~500 images with the cv2.calibrateCamera() method. I also tried with ~150 images and obtained very similar results.
You certainly could change the initialisation to reduce the solution space, but I would make sure the camera calibration is good (capture an image of an object with straight sides and make sure your calibration values are undistorting as expected) before venturing down that path.

laxnpander, 2017-01-26:
Update: I am actually quite an idiot. My calibration was done using full sized images while I fed ORB Slam resized images, which can't obviously work. Interesting enough he was able to initialize one time under these circumstances. I realized my mistake and now it works like a charme.
Thanks for your help, as a orb slam noob it was actually good to see, that it should definitely work in a scenario like mine!

SamuelDudley, 2017-01-26:
Yep that will do it! I'm still interested in seeing a video of your data set if you would like to share :)

laxnpander, 2017-01-26:
[https://youtu.be/TwpKRbQmyA4](Here you go!) There is still a lot of potential in parameter variation, we started a few days ago using orb slam. Especially running the camera with 2-3 FPS will make a huge difference I guess. But already a very powerfull tool out of the box.
To me however it seems a little bit magical at the moment. As soon as I increase the number of features the results are getting worse. Sometimes he already loses track after the first leg. Is there a special technique to identify best settings? I guess it's somehow a compromise between processing power, FPS and resolution?

shridharkini6, 2018-02-02:
@laxnpander can we premultiply to our intrinsic matrix when images are resized?
If we resized the image by half then we can multiply our intrinsic matrix by [ [0.5 0 0] [0 0.5 0] [0 0 1] ]*IntrinscMatrix_3x3_
Do you think this will work?
Thanks

laxnpander, 2018-02-03:
Yes, you can do that. But be aware not to scale the distortion coefficients, they remain fixed.