raulmur/ORB_SLAM2 #241: OpenCV Error

stoneee, 2017-01-25:
ubuntu 16.04 with ROS
my camera's topic is /camera/image_raw, my camera is working and i can see the image on rviz, but when I run Mono there is something wrong with opencv. I don't know why , pls help me .
[image]
[image]
[image]
[image]
[image]
[image]
[image]

stoneee, 2017-01-25:
[image]

hari0920, 2017-01-25:
I had a similar issue on 14.04 with Indigo. I solved it by not using a separate install of Opencv. Install only ROS and use the version that it comes with.
Good Luck!