raulmur/ORB_SLAM2 #242: Raspberry PI 3, Ubuntu 16.04 (Mate ) build error

chaowu2009, 2017-01-28:
32GB SD card. Encountered the following error while building the solution: 
In file included from /usr/local/include/eigen3/Eigen/Core:340:0,
                 from /usr/local/include/pangolin/gl/gl.h:39,
                 from /usr/local/include/pangolin/pangolin.h:33,
                 from /home/cwu/project/ORB_SLAM2/include/MapDrawer.h:27,
                 from /home/cwu/project/ORB_SLAM2/include/Viewer.h:26,
                 from /home/cwu/project/ORB_SLAM2/include/Tracking.h:28,
                 from /home/cwu/project/ORB_SLAM2/include/LocalMapping.h:27,
                 from /home/cwu/project/ORB_SLAM2/include/LoopClosing.h:25,
                 from /home/cwu/project/ORB_SLAM2/include/Optimizer.h:27,
                 from /home/cwu/project/ORB_SLAM2/src/Optimizer.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
**c++: internal compiler error: Killed (program cc1plus)**

goldbattle, 2017-01-30:
This is caused by the c++ compiler running out of memory. Run this on a program with large memory. There might be a compiler flag you can specify to build on low memory system, but you can try adding a `-j1` to your make command.

poine, 2017-01-30:
and add swap to your pi

Kirosoft, 2017-01-30:
something like this:

sudo fallocate /var/swapfile -l 1G
sudo chmod 600 /var/swapfile
sudo mkswap /var/swapfile
sudo swapon /var/swapfile

fnoop, 2017-03-05:
Same problem compiling on joule.  The build.sh script tries to always build orbslam2 with -j with no parameter, which is suicide on any system with limited memory.  Raspberry and joule both have quad cores so can parallelize compilation but have limited memory (if you can call 4gb limited, in the case of the joule), and each compilation process takes a lot of memory.
So change build.sh and alter each make -j line to make -j <x> where <x> is a sane value like 1 or 2.

oguzhankarabulut, 2017-03-13:
After the add swap you can see your swap with "free -m" and then build orbslam2 step by step. Do not run directly ./build.sh if you open cmake files you will see steps. These steps like these : 
```
cd Thirdparty/DBoW2
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make

cd ../../g2o
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make

cd ../../..
cd Vocabulary
tar -xf ORBvoc.txt.tar.gz

cd ..
mkdir build
cd build
cmake .. -DCMAKE_BUILD_TYPE=Release
make
```