raulmur/ORB_SLAM2 #243: Unity3D native plugin

roscopecoltran, 2017-01-29:
Hi guys,

Hope you are all well !

I was wondering if there is any plan to wrap ORB SLAM2 as native plugin c++ ([managed or un-managed](http://www.alanzucconi.com/2015/10/11/how-to-write-native-plugins-for-unity/)) for Unity3D. That would be awesome as there is plenty of native camera plugins available for iOS and Android, and would make development easier and faster ?

Have a good week !

Cheers,
Richard

roscopecoltran, 2017-01-30:
Hi,

Hope you are all well and thanks for the link ! :-)

To give you a better overview, I tried to wrap it into a static/shared library throught cmake with [hunter](https://github.com/ruslo/hunter) and [polly](https://github.com/ruslo/polly). You can check a cross-platform project with **OpenCV3 + OGLES_GPU**, using the **sniper/polly**, at this repo url https://github.com/headupinclouds/gatherer. 

Hunter makes it easier to[ build dependencies](https://github.com/ruslo/hunter/blob/master/cmake/projects/OpenCV/hunter.cmake#L216) per version and  Polly to setup the proper toolchain for each platforms; https://github.com/hunter-packages.

Ideally, it would work like this example from OpenCVForUnity https://github.com/EnoxSoftware/OpenCVForUnity/blob/master/OpenCVForUnity/Samples/ArUcoSample/ArUcoTexture2DSample.cs#L155

## **Unity Workflow**

1. Instantiate a detector object (using System.Runtime.InteropServices) 
2. for each frame updates, process the input matrix (provided by OpenCVForUnity)
3. process the returned transform values and apply them to a set of game objects.

## **Unity Odometry plugin wrapping**
1. Generate a shared/static library for several platforms from CMakeLists (Hunter + Polly) with a script (eg. [Mixpanel native plugin for Android, iOS, OSX, Windows and Linux](https://github.com/mixpanel/mixpanel-unity/blob/master/native/build/build_all.py#L16))
   a. would provided a shared library (*.so) for Android, Unity Editor
   b. would provided a static library (*.a) for MacOSX and iOS, and the bundle files
   c. would provided all DLLs for windows platforms
2. Generate wrapper around ORB_SLAM2 native detector methods and functions

Ps. I wanted to try https://github.com/mono/CppSharp and https://github.com/mono/embeddinator-4000 to make a kind of semi-automated wrapper  for cpp native plugins.

With that we could build a project for any kind of platform and have something more easy to develop, I think. Maybe, I am wrong...

Any feedback is welcomed :-)

Have a good day !

Cheers,
Richard

Martin20150405, 2017-01-30:
@roscopecoltran  
I seem to understand what you are trying to do. 
It's not very hard to build a native-plugin, the real problem is the performance of ORB_SLAM2 ,which is quiet unsatisfying on mobile devices(4-8 FPS on Android, ~10 FPS on IPhone if you don't do any optimizations).
Using SLAM to build a cross-platform AR application is not an urgent task for me.  And I'm not very familiar with Unity. So I will upload CMakeLists, DLL compiled for unity ,executable demo for windows and some SLAM AR codes on Unity3D (still have some bugs) later .  You can check it (I don't know whether it will help)
I may continue on my work about 2 weeks later. If you have any problems or feedbacks, please contact me. **Besides, a pull request is highly welcomed.**
BTW, sorry for my English...

roscopecoltran, 2017-01-30:
Are u french ?

Martin20150405, 2017-01-30:
........秒回啊，吓死我了

roscopecoltran, 2017-01-30:
martin sounds french, sorry about that :-)

Just to share some examples of ORB SLAM2 + Unity, I saw on youtube:
https://www.youtube.com/watch?v=rnODCkjPtq4
https://www.youtube.com/watch?v=XVhdpCmyQtw
https://www.youtube.com/watch?v=qQ9tkOiPAhQ

Martin20150405, 2017-01-30:
@roscopecoltran 
That's exactly what I'm working on. 
I saw these videos before, because he is my schoolmate and I can't get the source code...... :D

roscopecoltran, 2017-01-30:
He made an awesome job :-) And we have exchanged a quick email today

AhmadBehzadi, 2017-02-11:
@roscopecoltran, @Martin20150405 
Hi,
Got any result with that effort?

I'm trying to add a simple AR text in [this ](https://github.com/ygx2011/ORB_SLAM-IOS)iOS sample, but I'm very new to that and I don't know how to detect position (XYZ) of the real world object when user touches some where on screen(X,Y of mobile screen).
Any idea?

carlosfarinhas, 2017-07-13:
@roscopecoltran any news about your development on unity 3d native plugin, like gaoxiang?

carlosfarinhas, 2017-07-13:
I was able to build on windows with VS15, livecam/webcam, and ORBvoc.bin! 
But now, the problem is wrap ORB SLAM2 as a cross-platform native plugin c++ (managed or un-managed) for Unity3D...

carlosfarinhas, 2017-07-13:
And I have seen already the stereo version of it!

egemenertugrul, 2021-02-09:
Hey all, I just wanted to add that a GitHub user: [@Martin20150405](https://github.com/Martin20150405) actually implemented a .dll wrapper for Unity, he had the second comment in this thread, but he removed his repository and now his comment after I politely asked him to republish his repository. So, here's our conversation: 

https://ibb.co/VVT87yJ

and video: https://www.youtube.com/watch?v=q5uV02-Igpo (I'll reupload in case it gets down).