raulmur/ORB_SLAM2 #245: Changing System.h members to protected

laxnpander, 2017-01-30:
Hey ho,

currently I am integrating orb slam in my own use case using a wrapper class. I think it might be an option to change System.h member variables to protected permanently to allow derived classes to access parameters like calibration matrix and distortions. That way people like me won't need to change source code to extend functonalities. 
What do you think?

Best regards

Alex

AlejandroSilvestri, 2017-01-31:
Hi, I assume you mean changing private members to protected.

It doesn't seem a good idea changing them all.  Perhaps you should pick which one you need and why

I myself am messing with the code, changing many members to public to easy coding.  It doesn't mean it is a good idea to do it for everyone.

caselitz, 2017-01-31:
I think changing private to protected is a good idea. I also did it to inherit from System...

laxnpander, 2017-01-31:
I found it difficult to place my own code, because calibration reading is hardcoded into the library and I dont really want to read it twice. Don't get me wrong, sure it is easy to change for a developer. But as orb slam progresses there should definitely be a consideration of how to allow easy access for use cases

closed 2017-02-25 by laxnpander