raulmur/ORB_SLAM2 #250: Fail in building with ROS

Kanelankai, 2017-02-03:
Hello, everyone!
I have succeeded in building and testing it with monocular examples.
Now I want to run it with ROS examples.
However, when I built the nodes for mono,
errors occurred after executing build_ros.sh.
The error information is as follows:
/////////////////////////////////////////////////////////////////////////////////////////////
Scanning dependencies of target Mono
[ 20%] Building CXX object CMakeFiles/Mono.dir/src/ros_mono.cc.o
In file included from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41:0,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/LoopClosing.h:34,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/LocalMapping.h:26,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/Tracking.h:31,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/System.h:29,
                 from /home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_mono.cc:32:
/home/dars/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/Mono.dir/src/ros_mono.cc.o] Error 1
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make: *** [all] Error 2
//////////////////////////////////////////////////////////////////////////////////////
Does anyone know what the problem is? 
Thank you very much!

pisuto, 2017-02-12:
why i can't find build_ros.sh?
Did I lose something?

Kanelankai, 2017-02-12:
The build_ros.sh is in the ORB_SLAM2 file.
It has the same path with build.sh.

pisuto, 2017-02-12:
I find it.
But I met the same problem when executing it.
compilation terminated.
make[2]: *** [CMakeFiles/Mono.dir/src/ros_mono.cc.o] 错误 1
make[1]: *** [CMakeFiles/Mono.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....
make[2]: *** [CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o] 错误 1
make[1]: *** [CMakeFiles/MonoAR.dir/all] 错误 2
make[2]: *** [CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o] 错误 1
make[1]: *** [CMakeFiles/RGBD.dir/all] 错误 2
make[2]: *** [CMakeFiles/Stereo.dir/src/ros_stereo.cc.o] 错误 1
make[1]: *** [CMakeFiles/Stereo.dir/all] 错误 2
make: *** [all] 错误 2

Kanelankai, 2017-02-12:
I can't confirm if your problem is the same with me. In my case, "fatal error: Eigen/core :no such file or directory" is why this problem happens. Some blogs say that since orbslam2 uses Eigen 3, you can add a link to the path of Eigen 3/Eigen/core to solve this problem. I can't remember the command of adding the link. If you google "fatal error: Eigen/core :no such file or directory", I think you can find the same blog. Good luck!

发自我的 iPhone

pisuto, 2017-02-12:
thx.I will try.来自我的华为手机-------- 原始邮件 --------主题：Re: [raulmur/ORB_SLAM2] Fail in building with ROS (#250)发件人：Kanelankai 收件人：raulmur/ORB_SLAM2 抄送：pisuto ,Comment I can't confirm if your problem is the same with me. In my case, "fatal error: Eigen/core :no such file or directory" is why this problem happens. Some blogs say that since orbslam2 uses Eigen 3, you can add a link to the path of Eigen 3/Eigen/core to solve this problem. I can't remember the command of adding the link. If you google "fatal error: Eigen/core :no such file or directory", I think you can find the same blog. Good luck!

发自我的 iPhone

pisuto, 2017-02-13:
I find one.But I dont't know where I should change the code.
fatal error: Eigen/Core: No such file or directory - ROS Answers: Open Source Q&A Forum  http://answers.ros.org/question/44140/fatal-error-eigencore-no-such-file-or-directory/

Kanelankai, 2017-02-13:
I solved it by executing the following: sudo ln -s /usr/include/eigen3/Eigen /usr/local/include/Eigen
You can see the details from the following URL:
http://askubuntu.com/questions/491067/eigen-installation-seemed-to-work-but-i-still-cant-make-eigen-work

pisuto, 2017-02-13:
Thank you!
I solved my problem by executing the code you provide!

pisuto, 2017-02-13:
Sorry,I met another problem.Can you help me?
When I executed the code:rosrun ORB_SLAM2 MonoAR PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE
I met:
[ERROR] [1486956228.917143261]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...

Kanelankai, 2017-02-13:
Maybe you didn't launch your monocular camera or didn't run a roscore. I'm sorry but I'm not very familiar with this kind of problems.....

On 2017/02/13 12:26, pisuto wrote:

Sorry,I met another problem.Can you help me?
When I executed the code:rosrun ORB_SLAM2 MonoAR PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE
I met:
[ERROR] [1486956228.917143261]: [registerPublisher] Failed to contact master at [localhost:11311]. Retrying...

pisuto, 2017-02-14:
Sorry,I think I may bother you.
Now I can execute the example,but when executing Mono,I see a black window lining WAITING FOR IMAGES.
I want to know how to see the images.I try to use usb_cam,but I can't find a correct method.
Can you give me some suggestions or tutorials?

Kanelankai, 2017-02-14:
That's OK. Please don't mind.

Can you acquire the rgb image through your usb camera without using ORB-SLAM2?

If you cannot, I think you'd better check whether the driver of your camera has been installed or not.

Otherwise, I found that you can only use a Kinect or Xtion in RGBD mode, or an ordinary usb camera in mono mode due to the default settings. Since I haven't seen the contents of the code I have no idea about the details of these settings.

I succeeded in SLAM with ROS in RGBD mode with Xtion camera, When I executed it in Mono mode with an ordinary usb camera, I could only obtain images from the camera but failed to initialize. I haven't fixed this problem.

On 2017/02/14 11:36, pisuto wrote:

Sorry,I think I may bother you.
Now I can execute the example,but when executing Mono,I see a black window lining WAITING FOR IMAGES.
I want to know how to see the images.I try to use usb_cam,but I can't find a correct method.
Can you give me some suggestions or tutorials?

pisuto, 2017-02-14:
I execute:rosrun usb_cam usb_cam_node
I can success the camera without ORB_SLAM2.
But when using ORB_SLAM2,it shows WAITING FOR IMAGES.
I don't know "a monocular input from topic /camera/image_raw run node ORB_SLAM2/Mono"means.
I think my problem occur because of this.
Can you help me?

Kanelankai, 2017-02-14:
When you run the monocular examples, did you replace "PATH_TO_VOCABULARY" and "PATH_TO SETTINGS_FILE" with your own information? If you had already done it when running the codes, I'm sorry but I have no idea either.

On 2017/02/14 12:29, pisuto wrote:

I execute:rosrun usb_cam usb_cam_node
I can success the camera without ORB_SLAM2.
But when using ORB_SLAM2,it shows WAITING FOR IMAGES.
I don't know "a monocular input from topic /camera/image_raw run node ORB_SLAM2/Mono"means.
I think my problem occur because of this.
Can you help me?

pisuto, 2017-02-14:
Yes,I have already replaced these.I may try other method.
I am not familiar with ROS.
So I think I should learn its  tutorials first.
Thank you again.

pisuto, 2017-02-14:
I solved my problem.The original location which images are got from:/camera/image_raw doesn't match with the location which images are saved by usb camera.
I made them same.Then done.

wertig0n, 2017-02-20:
Hi!

Sorry for hijacking but I have a similar problem. First it would not find the eigen sources, but doing the symlink trick mentioned earlier in the thread worked for that.

Now I have problem number 2, which is that the linker will not find the opencv libraries for some unknown reason. I'm running Ubuntu 16.10 with ROS kinetic. I could compile everything fine before I installed ROS, but after I installed ROS I get this (only showing most relevant lines):

[ 81%] Linking CXX executable ../Examples/Stereo/stereo_kitti
CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o: In function `main':
mono_tum.cc:(.text.startup+0x504): undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'

Probably a very simple fix as well, but I do not know which line to alter.