raulmur/ORB_SLAM2 #253: Comparing 6DOF results and ground truth of EuRoC dataset

Oranjo, 2017-02-07:
Hi

I'm sorry to ask a basic question, but How can I compare the result(CameraTrajectory.txt) and the ground truth in EuRoC dataset? I'm currently trying V101, V201.
The position result shows similar pattern to the ground truth, but can't be directly compared with. It seems to require transformation matrix.
The camera pose result is quite different from the ground truth. I transformed quaternion to Euler angle and It does not seem to match the reference; especially roll and pitch angles are not.

How can I make 6DOF error graph?

Thanks.

closed 2017-03-05 by Oranjo

pigbreeder, 2017-06-21:
Hi @Oranjo .
Do you solve the problem?  
Do you know how to compare the result and the ground truth in KITTI dataset?

Thanks.

Oranjo, 2017-06-21:
Hi.
I could compare the results with this(
https://github.com/raulmur/evaluate_ate_scale)
But the results are not the same as paper.
I referred to this issue that you've already checked:
https://github.com/raulmur/ORB_SLAM2/issues/270

Thanks.
ᐧ

-------------------------------------------------------------------------

석사과정 김연조 / 지능형항법제어연구실

건국대학교 항공우주정보시스템공학과

서울특별시 광진구 능동로 120  건국대학교 신공학관 502호

Tel : 02-452-4176, H.P : 010-8851-2610

pigbreeder, 2017-06-22:
Thank you for your kind reply.
But when I use the command(./evaluate_ate_scale.py CameraTrajectory.txt V1_01_easy/mav0/state_groundtruth_estimate0/data.csv )
,it print Couldn't find matching timestamp pairs between groundtruth and estimated trajectory!
Where did i make mistake?

Oranjo, 2017-06-22:
Hi

I followed the instruction of original tool as the readme file says. (
http://vision.in.tum.de/data/datasets/rgbd-dataset/tools#evaluation)

I think it can be helpful for you. I'm sorry that I can't tell you in
detail cause I'm quite busy recently....

Best regards.

ᐧ

-------------------------------------------------------------------------

석사과정 김연조 / 지능형항법제어연구실

건국대학교 항공우주정보시스템공학과

서울특별시 광진구 능동로 120  건국대학교 신공학관 502호

Tel : 02-452-4176, H.P : 010-8851-2610