raulmur/ORB_SLAM2 #254: makefile: 83: recipe for target 'all' failed, trying to build ORB-SLAM2

ucsdxiaoyuzhou, 2017-02-09:
System information (version)

    OpenCV => :3.2
    Operating System / Platform => :ubuntu 16.04
    Compiler => :

Detailed description

I am following the steps from README.txt to build ORB_SLAM2 on my ubuntu. I am running ubuntu on virtual machine, and I allocated 6GB RAM for it.
I think I did all dependencies, and all other requirements, and got to the last step:
which is to build ORB-SLAM2 library and TUM/KITTI examples, I used following commands:
cd ORB_SLAM2
chmod +x build.sh
./build.sh

and there are some errors so I am not able to build it successfully. The error is as following:
[image]
[image]
[image]
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xiaoyu/Desktop/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xiaoyu/Desktop/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xiaoyu/Desktop/ORB_SLAM2/build
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
[  6%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
[  9%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
[ 12%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
[ 15%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o
In file included from /home/xiaoyu/Desktop/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/Tracking.h:31,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/Viewer.h:25,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/src/Viewer.cc:21:
/home/xiaoyu/Desktop/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
......
.....
......
.....
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
In file included from /home/xiaoyu/Desktop/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35:0,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc:21:
/home/xiaoyu/Desktop/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_binary_edge.h:60:80: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Dj, Di>, Matrix<double, Dj, Di>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockTransposedType;
                                                                                ^
In file included from /usr/local/include/eigen3/Eigen/Core:340:0,
                 from /home/xiaoyu/Pangolin/include/pangolin/gl/gl.h:39,
                 from /home/xiaoyu/Pangolin/include/pangolin/pangolin.h:33,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/MapDrawer.h:27,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/Viewer.h:26,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/Tracking.h:28,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/LocalMapping.h:27,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/include/LoopClosing.h:25,
                 from /home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackStereo(const cv::Mat&, const cv::Mat&, const double&)’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc:134:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackRGBD(const cv::Mat&, const cv::Mat&, const double&)’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc:185:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc: In member function ‘cv::Mat ORB_SLAM2::System::TrackMonocular(const cv::Mat&, const double&)’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc:236:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc: In member function ‘void ORB_SLAM2::System::Shutdown()’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc:309:24: error: ‘usleep’ was not declared in this scope
             usleep(5000);
                        ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/System.cc:315:20: error: ‘usleep’ was not declared in this scope
         usleep(5000);
                    ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc: In member function ‘void ORB_SLAM2::LoopClosing::Run()’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc:84:20: error: ‘usleep’ was not declared in this scope
         usleep(5000);
                    ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc: In member function ‘void ORB_SLAM2::LoopClosing::CorrectLoop()’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc:428:20: error: ‘usleep’ was not declared in this scope
         usleep(1000);
                    ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/Tracking.cc: In member function ‘void ORB_SLAM2::Tracking::Reset()’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/Tracking.cc:1512:24: error: ‘usleep’ was not declared in this scope
             usleep(3000);
                        ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc: In member function ‘void ORB_SLAM2::LoopClosing::RequestReset()’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc:630:20: error: ‘usleep’ was not declared in this scope
         usleep(5000);
                    ^
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc: In member function ‘void ORB_SLAM2::LoopClosing::RunGlobalBundleAdjustment(long unsigned int)’:
/home/xiaoyu/Desktop/ORB_SLAM2/src/LoopClosing.cc:670:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
CMakeFiles/ORB_SLAM2.dir/build.make:110: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/ORB_SLAM2.dir/build.make:494: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:86: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:134: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
CMakeFiles/ORB_SLAM2.dir/build.make:62: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/System.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

I really spent lots of time on it, and I have to finish installing it ASAP, so please help me out! I really appreciate you guys help!!!

ucsdxiaoyuzhou, 2017-02-09:
it's solved simply by adding 
#include <unistd.h>
in those error files

arntanguy, 2017-02-14:
I proposed in PR #260 to replace usleep with the now C++11 standard `std::this_thread::sleep_for` to permanently fix this issue.

vacky11, 2017-03-13:
@ucsdxiaoyuzhou thanks for the solution.
worked for me

riccitensor, 2017-12-11:
Perhaps related -- I am having this issue:
```

In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Tracking.h:31,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Viewer.h:25,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/src/Viewer.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
make[2]: *** No rule to make target '../Pangolin/build/src/libpangolin.so', needed by '../lib/libORB_SLAM2.so'.  Stop.
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

vacky11, 2017-12-11:
@riccitensor Can you post the entire terminal output with all the errors.

riccitensor, 2017-12-11:
```
tfuser@appliance0:~/Desktop/Development/ORB_SLAM2$ ./build.sh 
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tfuser/Desktop/Development/ORB_SLAM2/build
Scanning dependencies of target ORB_SLAM2
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
In file included from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Tracking.h:31,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Viewer.h:25,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/src/Viewer.cc:21:
/home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  Hes
                                                                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Tracking.h:31,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Viewer.h:25,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/src/Viewer.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
In file included from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Tracking.h:31,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Viewer.h:25,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/src/Viewer.cc:21:
/home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  Hes
                                                                          ^
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Tracking.h:31,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/include/Viewer.h:25,
[... 1032 lines cut ...]
                 from /home/tfuser/Desktop/Development/ORB_SLAM2/src/Viewer.cc:21:
[... cut ...]
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
make[2]: *** No rule to make target '../Pangolin/build/src/libpangolin.so', needed by '../lib/libORB_SLAM2.so'.  Stop.
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

riccitensor, 2017-12-11:
Pangolin was compiled already, but unfortunately not well enough. I solved it using #22

vacky11, 2017-12-11:
LOL!
In this digital world it is either 1 or 0, Yes or No, Compiled or Not complied. And I believe that if it is not compiled "well enough" it means that it is not complied.
But anyway, good to know that the solution provided in that issue helped you. Cheers!

riccitensor, 2017-12-11:
It was compiled, but did not work. The instructions in the https://github.com/raulmur/ORB_SLAM2/issues/22 helped. Thanks!

pedrobagatin, 2019-10-07:
> it's solved simply by adding
> #include <unistd.h>
> in those error files

Which error files?

furkancaglayan, 2019-10-07:
> > it's solved simply by adding
> > #include <unistd.h>
> > in those error files
> 
> We need to include it in all the files ?

LocalMapping.h
LoopClosing.h
System.h
Tracking.h
Viewer.h

wangzhuang156174330, 2021-02-17:
> it's solved simply by adding
> #include <unistd.h>
> in those error files

can you teach me in a example?

mfp0610, 2021-07-14:
https://github.com/mfp0610/ORB_SLAM2
You can refer to my fork, I have solved the problem.

HaochengZheng, 2021-12-06:
I added #include <unistd.h> to all possible positions and changed typedef map<KeyFrame*,g2o::Sim3,std::less<KeyFrame*>,
        Eigen::aligned_allocator<std::pair<const KeyFrame*, g2o::Sim3> > > KeyFrameAndPose; 

But I still have Makefile 83 recipe for target 'all' failed. That's the only error. Plz help

mfp0610, 2021-12-06:
> I added #include <unistd.h&gt; to all possible positions and changed typedef map<KeyFrame*,g2o::Sim3,std::less<KeyFrame*&gt;,
>  Eigen::aligned_allocator<std::pair<const KeyFrame*, g2o::Sim3&gt; &gt; &gt; KeyFrameAndPose;

To whom I may concerned :
Well I'd like to help but I am preparing for my final exam, may I reply your some time next week? 
Sincerely Meng.

HaochengZheng, 2021-12-06:
> To whom I may concerned : Well I'd like to help but I am preparing for my final exam, may I reply your some time next week?  Sincerely Meng.
> […](#)

sure thanks for your quick reply