raulmur/ORB_SLAM2 #256: Disabling loop closing and relocalization for using ORB_SLAM as VO algorithm

kerolex, 2017-02-09:
I would like to run ORB-SLAM in monocular mode as a simple VO algorithm (i.e. without loop closing and relocalization functionalities) for some evaluations.

I commented the two lines regarding initializing the Loop closing thread  and launch in System.cc. 
I also commented all lines with `bOK = Relocalization();`
Please see below the file with the changes:
[slam2vo_patch.txt](https://github.com/raulmur/ORB_SLAM2/files/764648/slam2vo_patch.txt)

However, when running the compiled algorithm on the first 11 KITTI sequences, "ORB-VO" does not work. For example, I got segmentation fault or aborted. I believe, the algorithm without thesxe two functionalities is not able to initialize or is calling something else that I should comment.

Can anybody tell me if I correctly disabled the loop closing and relocalization functionalities to have a simple VO algorithm, please?

AlejandroSilvestri, 2017-02-17:
ORB-SLAM initialize only once.  Then, when tracking is lost, it turn into relocalization mode.  If you disable it, the system will be lost forever, no attempting to reinitialize.  You can replace Relocalization() with a reinitialization.

Loop closing runs in a separate thread, you can disable it, but you should modify the code, there's no off switch.  For example, change the code so it stops providing new keyframes to loop closing processing FIFO queue.

As I understand VO, it is SLAM without mapping.  In order to get a "real" VO you should erase the map, perhaps working only with local map.  But this can be a huge work.

I think you already know this, but I write it down for others: OBR-SLAM2 has a "VO" mode, but it isn't a complete VO system, it is only a intermediate state between tracking and lost.

kerolex, 2017-03-01:
Thank you very much for your reply. 

I got what you are suggesting and I will try to see if I can make the further changes. In Direct Sparse Odometry paper, they mentioned some comparisons with ORB-SLAM, but they deactivated the two functionalities of Relocalization and Loop Closing. I was thinking it could be simpler, but it might require more work.

In addition to your comment about VO, it is kind of true you have to remove the map, but only in the case of VO with 2D-2D motion estimation. Nowadays, VO differentiates from Visual SLAM only in terms of capabilities like loop closure and relocalization. Indeed, VO estimates the 3D map but "remembering" only the most recent seen 3D points and performing within an "infinite" corridor. Thus, VO aims to estimate the trajectory of the camera as accurate as possible using only the most recent poses and 3D maps, whereas Visual SLAM aims to build a global and consistent map avoiding and reducing drift by recognizing previous visited locations. As good readings (at least they were for me), I would suggest the Visual Odometry tutorial by Davide Scaramuzza as well as the recent survey on SLAM: ["Past, Present, and Future of Simultaneous Localization and Mapping: Toward the Robust-Perception Age"](http://rpg.ifi.uzh.ch/docs/TRO16_cadena.pdf), in which Scaramuzza is again one of the authors.

I hope this can be useful for everyone.

AlejandroSilvestri, 2017-03-02:
@kerolex, you are right.  "Toward robust perception age" is a good reading, worth it.  I took VO definition from there.

As you pointed out, VO has local mapping.  Depending on your trajectory, it will make not difference with full map if you don't come back to already mapped places.  If you do, then SLAM will have more accuracy than VO, and if you want to do a real VO with ORB-SLAM2, you'll have to limit point and keyframe search to local map only.

zhangzichao, 2017-04-20:
For me, a quick hack is to let the corresponding functions always return false...
For loop closure:
https://github.com/raulmur/ORB_SLAM2/blob/master/src/LoopClosing.cc#L103
For relocalization:
https://github.com/raulmur/ORB_SLAM2/blob/master/src/Tracking.cc#L1341

It will still do some extra work, but at least behaves like a pure VO.

kauevestena, 2018-01-08:
Thank you, zhangzichao! I was aiming for the same stuff.
"It will still do some extra work" what kind of stuff?

Now I want just to save all the relative computed poses (all graph edges), the edge between 0 and 1, 0 and 2, 0 and 3...

How to achieve that?

kerolex, 2018-05-08:
For those that are interests in disabling the loop closure:

At the [Robotics and Perception Group GitHub repository of ORB-SLAM](https://github.com/uzh-rpg/ORB_SLAM2/blob/master/orb_slam_2/include/ORB_SLAM2/System.h), they made a modification adding the function `void ShutDownLoopClosure();` that allows to switch off the Loop Closure thread, making the system working as a Visual Odometry approach (I haven't tested it yet, but it might be useful).

UannaFF, 2019-04-09:
Hello, is there any improvement with respect to this? Has anyone found a solution? I was using ORBSLAM for VO of a shelve, so it's basically only translation. and when I have a few images it's ok. The accuracy is very good, but when I pass more and more images, the map starts curving when it shouldn't and I get a lot of rotation errors.

AlbertoJaenal, 2019-07-19:
Hello everyone, 
I have come up with a modification to disable LC and keep using ORB_SLAM2 as a VO algorithm. I don't know if this is  what you need but here I attach the main changes in the next .log file [DisableLoopClosing.log].(https://github.com/raulmur/ORB_SLAM2/files/3411587/DisableLoopClosing.log)

Fundamentally, I have commented all the lines in which the different classes call the LoopClosing object or thread:

- **Tracking.cc:** Line 1522 (https://github.com/raulmur/ORB_SLAM2/blob/master/src/Tracking.cc#L1522)
- **LocalMapping.cc:** Line 87 (https://github.com/raulmur/ORB_SLAM2/blob/master/src/LocalMapping.cc#L87)
- **System.cc:**
- - Lines 94-95: (https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc#L94-L95)
- - Line 107: (https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc#L107)
- - Line 110: (https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc#L110)
- - Lines 112-113: (https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc#L112-L113)
- - Line 304: (https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc#L304)
- - Line 313: **Here I changed slightly the code** (https://github.com/raulmur/ORB_SLAM2/blob/master/src/System.cc#L313), and wrote while(!mpLocalMapper->isFinished())

Please note me if this don't work for you or this is not what you looked for.

dishank-b, 2022-02-03:
Can anyone confirm if the ^ above method works to turn off the loop closing?