raulmur/ORB_SLAM2 #258: segment error when testing TUM !!

EXing, 2017-02-10:
./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml /home/huangkun/ORB_SLAM2/Examples/RGB-D/rgbd_dataset_freiburg1_xyz Examples/RGB-D/associations/fr1_xyz.txt

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.09294

-------
Start processing sequence ...
Images in the sequence: 792

New map created with 833 points
-------

median tracking time: 0.0401444
mean tracking time: 0.0420971

Saving camera trajectory to CameraTrajectory.txt ...

trajectory saved!

Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!
Segment error

I'm using opencv 3.2. And to fix the 'usleep undefined' error, I have added  'unistd.h' to 'system.h' manually.

EXing, 2017-02-10:
it happens when 'return 0' in rgbd_tum.cc!

`    // Save camera trajectory
    SLAM.SaveTrajectoryTUM("CameraTrajectory.txt");
    SLAM.SaveKeyFrameTrajectoryTUM("KeyFrameTrajectory.txt");   

    return 0;`

mtee, 2017-02-14:
fyi, segmentation errors are hard to track down to a line. I would not rely on the debugging. You can try to insert a 
`cout << " step X performed " << endl;`
after critical lines in the function SaveKeyFrameTrajectoryTUM. Segmentation faults mostly happen when a variable has not been properly initialized or set and then you try to read from it.

EXing, 2017-02-16:
it happens after SaveKeyFrameTrajectoryTUM！

mtee, 2017-02-16:
it can't happen in the line "return 0;", so I advice you to look inside the SaveKeyFrameTrajectoryTUM function. There is probably some null pointer in it, which leads to the segmentation fault when it is accessed.

EXing, 2017-02-16:
but "cout << " step X performed " << endl;" have already run.

EXing, 2017-02-16:
is it possible the return address in stack broken?

nickxiang0306, 2022-07-15:
https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/452 maybe this issue could help you to fix this segfault bug. And I met the bug by when running orb-slam3.