raulmur/ORB_SLAM2 #271: can we use orb-slam2 as pose estimation with stereo camera for dense reconstruction?

dejunzhang, 2017-03-08:
Hi, does the accuracy of the pose from orb-slam with stereo camera is good enough for dense reconstruction? thanks a lot.

AlejandroSilvestri, 2017-03-18:
Well, it depends on what accuracy you are looking for.

Of course you can do dense reconstruction based on orb-slam2 pose estimation.  I believe you must try it and evaluate the results.

Orb-slam2, like all monocular SLAM, is subject to drifting, including scale drifting.  The drifting is dramatically lowered with closed loops.  So, I recommend closing large loops in you path.

Pose accuracy depends on camera calibration, distorsion and the stereo camera extrinsic parameters accuracy measurements.

paulpaul91, 2017-07-28:
can you show us a good stereo calibration?

AlejandroSilvestri, 2017-07-31:
@paulpaul91 , sorry, I can't.

ouerkakaChango, 2017-12-21:
I've tried similar idea with RGB-D Example,use the result keyFrame as pose estimation,the result is bad.
Hope to know have you guys found a good way to do dense reconstruction with ORB-SLAM2，thanks