raulmur/ORB_SLAM2 #272: Can't build ROS module

wertig0n, 2017-03-09:
Ubuntu 16.04, Kinetic Kame installed by packages.

ORB_SLAM2 builds fine, but, I get this error message when running ./build_ros.sh:

$ ./build_ros.sh 
Building ROS nodes
mkdir: cannot create directory ‘build’: File exists
[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is
  "/home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
See also "/home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeError.log".
[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py  /home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/kinetic/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/kinetic/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is
  "/home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2".  You should
  double-check your ROS_PACKAGE_PATH to ensure that packages are found in the
  correct precedence order.
Call Stack (most recent call first):
  /opt/ros/kinetic/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
See also "/home/per/Work/plugg/DVA439/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeError.log".
Makefile:694: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1

wertig0n, 2017-03-10:
Ok, after searching TFW some more and re-reading the installations, I realised what was wrong. I didn't have a properly configured ROS_PACKAGE_PATH.

I do prefer the more elegant solution of symlinking from the include dir however, like this:

`sudo ln -s /home/myuser/compile/ORB_SLAM2/Examples/ROS/ORB_SLAM2 /opt/ros/kinetic/share/ORB_SLAM2`

It is also possible to add the path to the ROS_PACKAGE_PATH variable, and if you do not have root access that is the way to go instead. But yeah.

Closing this, thanks for having patience with me. :)

closed 2017-03-10 by wertig0n

RahmanMifta, 2018-06-29:
I am getting this error message after trying to symlinking 

ln: failed to create symbolic link '/opt/ros/kinetic/share/ORB_SLAM2': File exists

chengji253, 2019-04-02:
I also meet the problem that ln: failed to create symbolic link '/opt/ros/kinetic/share/ORB_SLAM2': File exists. Please tell me how to solve it?