raulmur/ORB_SLAM2 #276: too many key frames!!!

EXing, 2017-03-12:
I'm using a calibratied kinect2 with orbslam2. but when I using it, even with a small move of 4cm, it produce  so many keyframes. why? BTW, the dense is ugly.

EXing, 2017-03-12:
the kinect2 input code:
while(1)
    {
		if(!running)
			return;
		
		if(!updateCloud)
		{
		  if(!ros::ok())
		  	return;
		}
		else
		{
			lock.lock();
			updateCloud = false;
		  	orbslam->TrackRGBD( color, depth, ros::Time::now().toSec() );
		  	lock.unlock();
		}
    }

void callback(const sensor_msgs::Image::ConstPtr imageColor, const sensor_msgs::Image::ConstPtr imageDepth,
                const sensor_msgs::CameraInfo::ConstPtr cameraInfoColor, const sensor_msgs::CameraInfo::ConstPtr cameraInfoDepth)
  {
    cv::Mat color, depth;
    
    readImage(imageColor, color);
    readImage(imageDepth, depth);
    //readCameraInfo(cameraInfoColor, cameraMatrixColor);
    //readCameraInfo(cameraInfoDepth, cameraMatrixDepth);

    // IR image input
    if(color.type() == CV_16U)
    {
      cv::Mat tmp;
      color.convertTo(tmp, CV_8U, 0.02);
      cv::cvtColor(tmp, color, CV_GRAY2BGR);
    }

	lock.lock();
    this->color = color;
    this->depth = depth;
    updateCloud = true;
    lock.unlock();
  }

EXing, 2017-03-12:
And when it's running, there will be some warning:
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()

EXing, 2017-03-12:
Camera Parameters: 
- fx: 1061.64
- fy: 1062.49
- cx: 956.825
- cy: 558.988
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 30
- color order: BGR (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 2.80001
init done
New map created with 680 points
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()

EXing, 2017-03-13:
it's because of the output of cameraMatrixColor and cameraMatrixDepth in kinect2_bridge of iai_kinect2 is different from calibration data! and I used the original calibration data as parameters! this cause too many keyframes!

EXing, 2017-03-13:
but 
virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()
I still don't know why!

EXing, 2017-03-13:
@raulmur please!!thank you!

mtee, 2017-03-23:
> it's because of the output of cameraMatrixColor and cameraMatrixDepth in kinect2_bridge of iai_kinect2 is different from calibration data! and I used the original calibration data as parameters! this cause too many keyframes!

this is probably the main issue here. Other issues come from that. Are you sure your RGB resolution is 1080p? It does not look like this on the screenshot you provided, but the calibration data is obviously for this resolution (1080p)

EXing, 2017-03-23:
@mtee kinect2 provide hd（1080p），qhd（540×960），ir（smaller）. I used qhd.

mtee, 2017-03-23:
@EXing then as I said, you have to use other calibration data, since yours is not suitable at all. Your principal point (cx, cy) is supposed to be somewhat in the centre of the image, but in your case it's actually outside the image:
cx: 956.825
cy: 558.988

It simply cannot work like that. You have to use camera parameters that were acquired with the resolution you use the algorithm with. Good luck!

EXing, 2017-03-23:
@mtee https://github.com/code-iai/iai_kinect2/issues/385 .will it affect camera.bf?

mtee, 2017-03-23:
sure, since qhd-calibration data will most likely have another value for camera.fx. 

camera.bf is just camera.fx * baseline.

EXing, 2017-03-23:
@mtee it helps a lot! thank you!

EXing, 2017-03-23:
@mtee I reset camera.bf, but
"virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()"
 still happens!

closed 2017-03-25 by EXing