raulmur/ORB_SLAM2 #277: SLAM MODE runs TOO SLOW

hnanhtuan, 2017-03-13:
Hi everyone,

I am trying to run SLAM on my own sequence of images (monocular). 
However, it takes too long to run a frame (~2 min). Note that the software run normally (fast ~0.25s) on the provided dataset. I also modified camera calibration and distortion coefficients.
The locations of displayed keyframes also seem to be incorrect.

Please help to give me some pieces of advice what should I do in this case to make the software run at real-time.

Thank you very much.

hnanhtuan, 2017-03-13:
The root cause is the time stamp. The software will sleep for a certain time to synchrony with the next frame. The way I define the time stamp is different that cause the problem.

`if(ttrack<T)
            usleep((T-ttrack)*1e6);`

closed 2017-03-13 by hnanhtuan

barloccia, 2019-09-18:
I have a similar problem. Changing only the rgb.txt values (the timestamps) is causing different behavior of ORBSLAM2. Any idea to how to define timestamps from video frames?