raulmur/ORB_SLAM2 #279: How to do SLAM with RaspberryPi PiCam and IMU Sensor ?

oguzhankarabulut, 2017-03-14:


tugyan, 2017-03-15:
We are currently implementing similar thing using ORB-SLAM for an UAV. There is a nice paper written by creators of ORB-SLAM that you can look at: https://arxiv.org/abs/1610.05949. However, do note that implementation of this paper might be too much for a senior project. Hence I would suggest that first take small steps by simply finding the scale between real world and ORB-SLAM world. Note that this will eliminate most of the benefits of ORB-SLAM since there are severe problems with directly integrating IMU and using it to find scale. After that you might try several filters, maybe averaging filter, median filter or EKF if it is possible. Currently we are using a fairly strong ground station computer hence expensive comhutations are no problem for us, but you need to consider many problems with computation since you are using a raspberry pi. But the most important thing is, ORB-SLAM is not entirely easy to understand, decide wisely if you really need it, if it is necessary or not. Most of the cases, an EKF SLAM algorithm with IMU integrated will be enough. If you really want to use ORB-SLAM, you will need to invest a huge time for it to understand and use it wisely. You can start by reading relevant issue entries in GitHub.

oguzhankarabulut, 2017-04-04:
Firstly many thanks for helpful suggestions, like you said almost all resources are showed it is not possible with just one raspberry pi due to computing problems. Do you know any example tutorial or resource about EKF slam ? @tugyan 
regards

tugyan, 2017-04-04:
Sure, you can check this project: https://github.com/ethz-asl/ethzasl_msf . It is implemented for ROS systems but i guess you can install ROS on your device. It also utilizes IMU data. For theoretical background you can check this paper: http://www.margaritachli.com/papers/IROS2013fusion_paper.pdf