raulmur/ORB_SLAM2 #284: Issue with bundle adjustment

Mieos, 2017-03-17:
I use ORB-SLAM2 in stereo-mode with rectified images coming from a stereo calibrated camera.

I tried to see if the bundle adjustment works properly by testing it with a plane surfaced quite well textured and by trying to do a loop in the camera poses in order to get this loop detected by orbslam2. According to the pointCloud obtained, it seems to work properly during a time and it looks like a plane but after a while, the drift accumulates and when I come back to the first camera position, the reprojected points do not match at all in the pointcloud and I get 2 distinct planes. I think at this point a loop should be detected and the camera poses graph should be corrected but it's not.. Even If I try to come back in area I have already explore, the graph never recover as it should.

I've observed that I've got a quite huge amount of keyframes that could be a source of problems for the bundle adjustment (I did not really understand how to properly reduce this amount). I guess it might have a link with this parameter **ThDepth** but I don't know how to properly set it.

During the exploration I've got some warning in the terminal:
`virtual int g2o::SparseOptimizer::optimize(int, bool): 0 vertices to optimize, maybe forgot to call initializeOptimization()`

It seems that some other issues already report this ( #211 and #276 ) and it might be why the bundle adjustment does not work properly in my case.

Maybe there is a way I can reduce the number of keyframes used?

Mieos, 2017-03-20:
Ok..
So I investigated a little and I've found different things:
* First the issue was not linked with the warnings I've got in the terminal. These warnings were caused by the different adjustments done in the Optimizer (Local and global bundle adjustments, pose estimation etc..)  when you don't have enough correspondences and I guess it just does nothing in that case so.. this warning in not that important.
* Second, I've checked and yes, the global bundle adjustment was NEVER called. In the loop closing class, all the different conditions was never met to detect properly a loop and to launch global adjustment
* I've changed the baseline * fx parameter to fit the kind of value you have in the kitti dataset (so everything that I will obtain should be rescale but the tracking problem stay the same) and .... the global bundle adjustment seems to work properly. So I guess there is some parameters to fix to properly detect loop and properly use global bundle adjustment.

Any ideas concerning those parameters? Maybe it's linked to <b>ThDepth</b>.

mtee, 2017-03-23:
You can sort of control the frequency with which keyframes are created with the camera.fps parameter. The more the fps number, the less keyframes are created per time unit.