raulmur/ORB_SLAM2 #287: Error in librealsense while building build_ros.sh

palliprasanth, 2017-03-19:
[ 20%] [ 40%] [ 60%] make[2]: *** make[2]: *** No rule to make target `/opt/ros/indigo/lib/librealsense.so', needed by `../Stereo'No rule to make target `/opt/ros/indigo/lib/librealsense.so', needed by `../Mono'.  Stop.
.  Stop.
make[2]: make[2]: *** Waiting for unfinished jobs....*** Waiting for unfinished jobs....

make[2]: *** No rule to make target `/opt/ros/indigo/lib/librealsense.so', needed by `../RGBD'.  Stop.
make[2]: *** Waiting for unfinished jobs....
Building CXX object CMakeFiles/Stereo.dir/src/ros_stereo.cc.o
Building CXX object CMakeFiles/RGBD.dir/src/ros_rgbd.cc.o
Building CXX object CMakeFiles/Mono.dir/src/ros_mono.cc.o
[ 80%] [100%] make[2]: *** No rule to make target `/opt/ros/indigo/lib/librealsense.so', needed by `../MonoAR'.  Stop.
make[2]: *** Waiting for unfinished jobs....
Building CXX object CMakeFiles/MonoAR.dir/src/AR/ViewerAR.cc.o
Building CXX object CMakeFiles/MonoAR.dir/src/AR/ros_mono_ar.cc.o
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/MonoAR.dir/all] Error 2
make[1]: *** [CMakeFiles/RGBD.dir/all] Error 2
make[1]: *** [CMakeFiles/Stereo.dir/all] Error 2
make: *** [all] Error 2

closed 2018-03-14 by palliprasanth

odinhr, 2018-11-12:
Hi @palliprasanth ,

Can you solve it? I have similar problems with librealsense and libusb.

Kanishk598, 2021-04-02:
Did anyone solve it? I am also getting the same error:
[image]

Kanishk598, 2021-04-02:
Coincidentally I solved this issue right after posting my problem, after spending almost two days with the problem and googling. 
What you need to do is to install librealsense ROS, which can be done using http://wiki.ros.org/librealsense