raulmur/ORB_SLAM2 #290: Performance problem in  mobile device

ningduyanghe, 2017-03-21:
Hi,I am planning to use orb_slam2 in my AR app,but the performance in mobile device is not adequate at all(<5fps).I test PTAM in the same device,but it perform better(>20fps).I read your papers but I am not good at algorithm,so...I just can not accept the results because I like your beautiful codes.

ningduyanghe, 2017-03-21:
btw,in ios and android,camera rotation can be read from motion sensors directly.

AlejandroSilvestri, 2017-03-25:
@ningduyanghe , if you wan't to integrate motions sensors, you can do it with Kalman filter or something like that.  Not so straightforward.

But you must know visual slam is far more accurate than motion sensors.

ORB-SLAM couldn't perform well on mobile.  The firs goal is to obtain an acceptable performance in tracking only.

ningduyanghe, 2017-03-27:
@AlejandroSilvestri You realy help me a lot!!!

closed 2017-03-27 by ningduyanghe

AlejandroSilvestri, 2017-03-27:
@ningduyanghe , my pleasure.