raulmur/ORB_SLAM2 #293: Rectified images as input

romdos, 2017-03-22:
Hi, all!
I'm trying to run ORB-SLAM2 with ROS Kinetic and Mono camera and the question is what type of image should we feed to this algorithm? 
Suppose I've calibrated my camera and I have usb_cam/image_raw and  [usb_cam/image_rect topics](http://wiki.ros.org/image_proc). What of them should I give to ORB-SLAM as camera/image_raw parameter? The parameter's name tell us that it shoud be a raw (uncalibrated) image most likely and I've noted that it accepted only usb_cam/image_raw topic. But still may be this question seems to be weird but is it redundant information of camera in Examples folder when we are using ROS? 
My question is also a little bit related on [this](http://answers.ros.org/question/257047/image_proc-vs-image_raw/).

AlejandroSilvestri, 2017-04-01:
Hi @romdos ,

First, calibration parameters involve intrinsic and undistortion parameters.  You won't need the latter if your images are rectified.  You always need intrinsic parameters (focal distance, principal point).

I only use mono.  If you provide your camera's undistortion parameters in orb-slam2 configuration file, you must feed raw images, because orb-slam2 will undistort points.

Some datasets provide rectified (ie undistorted) images.

Rectifying images is time consuming, and In a real time application it's a waste of time.  orb-slam2 only undistort keypoints, a much lighter work than undistort (ie rectify) the whole image.

I hope it helps.

romdos, 2017-04-03:
Thank you for your reply! If I give already rectified images to the SLAM,  should I set k1 k2 to zero in .yaml file? (I run it in ros).

AlejandroSilvestri, 2017-04-03:
@romdos , yes, set both radial distortion parameters to zero.

Grifo07, 2017-04-11:
Hi everyone!

How about the p1 and p2 distortion parameters? Should I also set them to zero in case I feed a rectified video stream to the ORB-SLAM2?

Thank you in advance

AlejandroSilvestri, 2017-04-13:
@Grifo07 
Yes, set them to zero.

adnion, 2017-06-19:
I would like to use "/camera/image_raw/compressed" with ORB_SLAM2. So I changed the topic in "ros_mono.cc" and recompiled it by running "build_ros.sh". However I only get a black image.

The camera node uses my custom calibration parameters. In the ORB_SLAM2 camera settings file I set all distortion parameters to zero, like @AlejandroSilvestri suggested.

If ORB_SLAM2 only supports raw image data then how would you deal with compressed images?

arb93, 2018-02-01:
I am currently providing the code with raw images. The .yaml file has the intrinsic properties and dist coeffs for both cameras in the stereo pair. What is the calib.txt file?

Is that used? What are the matrices P0, P1, P2 and P3 in the kitti stereo dataset?

Thanks.