raulmur/ORB_SLAM2 #294: terminate called after throwing an instance of 'std::bad_alloc'   what():  std::bad_alloc

suijingfeng, 2017-03-24:
hello, I'm a student from China, I successfully compile the ORB_SLAM2 as the read me, 
but I can not run it.
--------------------------------------------------------- 
Loading ORB Vocabulary. This could take a while...
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
Aborted (core dumped)
[image]
 the memory usage up to 99%, then core dumps, I don't know how to debug this, any helps? thanks.

romanomatthew23, 2017-04-18:
Hello, I had the same issue. After removing ORB_SLAM2, and re-downloading and compiling it was fixed for me. No idea why but it is working now.

l1va, 2017-10-12:
Had the same issue - found that i pass wrong path to Vocabulary. In such case should be an exception i think. (i run mono with ROS)

joebedard, 2018-05-16:
I had the same problem. I was passing the vocabulary directory instead of the vocabulary file. Correct usage (for example mono_tum) is:
./mono_tum path_to_vocabulary_file path_to_settings_file path_to_sequence_dir

closed 2018-05-18 by suijingfeng