raulmur/ORB_SLAM2 #301: Live Stream with ORB SLAM 2

Verru94, 2017-03-30:
Hi,
I'm new in ORB SLAM and this is my question: is possible to do live stream with my web cam without using ROS? I'm sorry if this is a stupid question but I'm not very good at this. Thanks!

tiejiang, 2017-03-31:
I am new here too, but i think it is possible to do live stream with your web camera, for , the source code is independent of the ros, and in the paper they said they will be provide open source code for ROS.

Verru94, 2017-03-31:
Ok but I have to change something in the code?

Verru94, 2017-03-31:
Sorry I meant: I have to create my program?

AlejandroSilvestri, 2017-04-01:
I did it, but I had to modify orb-slam2.

https://github.com/AlejandroSilvestri/os1

See main()'s main while loop.

Verru94, 2017-04-01:
Thank you very much

Jephinus, 2017-08-23:
Hi Alejandro, i am new to this. How can i compile and run your os1 folder as i need to build map by using webcam without ROS. Thank you.

AlejandroSilvestri, 2017-08-24:
@Jephinus , I'm answering your question on os1 repository.