raulmur/ORB_SLAM2 #305: Internal compiler error

hjjayakrishnan, 2017-04-08:
While executing build.sh, it started throwing up internal compiler errors.
My gcc version : gcc (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.4) 5.4.0 20160609
Ubuntu : 16.04
ROS: Kinetic

Error: 
```
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/ar_go_ws/src/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/ar_go_ws/src/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/odroid/ar_go_ws/src/ORB_SLAM2/build
[  3%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
[  6%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
[  9%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
[ 12%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o
[ 15%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/FrameDrawer.cc.o
[ 21%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Frame.cc.o
[ 25%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrame.cc.o
[ 18%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/MapDrawer.cc.o
[ 28%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/PnPsolver.cc.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/KeyFrameDatabase.cc.o
[ 34%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
[ 37%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/ORB_SLAM2.dir/build.make:470: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Initializer.cc.o] Error 4
make[2]: *** Waiting for unfinished jobs....
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
CMakeFiles/ORB_SLAM2.dir/build.make:86: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 4
```
And these errors keep going and I have to forcefully stop it, because it froze up my PC. Any ideas why this is happening? Thanks

hjjayakrishnan, 2017-04-08:
Hi, I managed to resolve the issue. It was a memory problem so I had to options in hand: either add more memory to swap space and then execute build.sh or remove the parallel make operation in the build file. The former is would help in faster compilation.

closed 2017-04-08 by hjjayakrishnan

meaninglessyl, 2018-02-11:
nice! i have caught in it for a whole night