raulmur/ORB_SLAM2 #306: 'Couldn't find executable'

hjjayakrishnan, 2017-04-09:
Hi, I'm trying to run ORB_SLAM2 on ubuntu 16.04, ROS Kinetic. But while trying to `rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUM1.yaml`

I'm getting 

`[rosrun] Couldn't find executable named RGBD below /home/odroid/ar_go_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2`
It compiled successfully without any errors. Any idea what might be happening?

a-jahani, 2017-04-09:
Go check in "/home/odroid/ar_go_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2" you should have an executable file named RGBD. If there isn't any, you should compile build_ros.sh again and make sure it will compile without error.

closed 2017-04-09 by hjjayakrishnan

liu-felix123, 2022-02-21:
> Go check in "/home/odroid/ar_go_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2" you should have an executable file named RGBD. If there isn't any, you should compile build_ros.sh again and make sure it will compile without error.

HI sir, i have compiled  build_ros.sh without error, but /src/ORB_SLAM2/Examples/ROS/ORB_SLAM2 didn't generate the executable file named RGBD,.what's problem in the process,give me some advise ,pls.
the build_ros.sh terminal text (no error info ) as below:

/catkin_ws/src/ORB_SLAM2$ ./build_ros.sh

Building ROS nodes
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:7 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:16 (cmake_policy):
  The OLD behavior for policy CMP0002 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:18 (cmake_policy):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:20 (cmake_policy):
  The OLD behavior for policy CMP0005 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

CMake Deprecation Warning at /opt/ros/melodic/share/ros/core/rosbuild/rosbuild.cmake:23 (cmake_policy):
  The OLD behavior for policy CMP0011 will be removed from a future version
  of CMake.

[... 25 lines cut ...]
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.29
-- BUILD_SHARED_LIBS is on
[rosbuild] using multiarch 'x86_64-linux-gnu' for finding Boost
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
[rosbuild] Including /opt/ros/melodic/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/melodic/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/melodic/share/rospy/rosbuild/rospy.cmake
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.1.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/lfh/catkin_ws/src/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build