raulmur/ORB_SLAM2 #307: beg for your help

freshn, 2017-04-09:
when i do ./build.sh ,it will stop at 55% and get into a cycle as following:
(I omit the repetition of the part because it is too much.The error appeared only once)

In file included from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35:0,
                 from /home/fresh/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/fresh/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/fresh/ORB_SLAM2/src/LocalMapping.cc:21:
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_binary_edge.h:60:80: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Dj, Di>, Matrix<double, Dj, Di>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockTransposedType;
                                                                                ^
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/fresh/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/fresh/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/fresh/ORB_SLAM2/src/LocalMapping.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
......(Repeat above)
In file included from /usr/local/include/eigen3/Eigen/Core:347:0,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/fresh/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/fresh/ORB_SLAM2/include/Optimizer.h:27,
                 from /home/fresh/ORB_SLAM2/src/Optimizer.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int, long int>]’:
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:804:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int, long int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:782:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>]’
/usr/local/include/eigen3/Eigen/src/Core/PlainObjectBase.h:710:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Derived = Eigen::Matrix<int, -1, 1>]’
/usr/local/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; _Scalar = int; int _Rows = -1; int _Cols = 1; int _Options = 0; int _MaxRows = -1; int _MaxCols = 1]’
/usr/local/include/eigen3/Eigen/src/Core/PermutationMatrix.h:365:17:   required from ‘Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>& Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>::operator=(const Eigen::PermutationBase<OtherDerived>&) [with Other = Eigen::PermutationMatrix<-1, -1, long int>; int SizeAtCompileTime = -1; int MaxSizeAtCompileTime = -1; _StorageIndex = int]’
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:68:18:   required from ‘void g2o::LinearSolverEigen<MatrixType>::CholeskyDecomposition::analyzePatternWithPermutation(g2o::LinearSolverEigen<MatrixType>::SparseMatrix&, const PermutationMatrix&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; g2o::LinearSolverEigen<MatrixType>::SparseMatrix = Eigen::SparseMatrix<double, 0>; g2o::LinearSolverEigen<MatrixType>::PermutationMatrix = Eigen::PermutationMatrix<-1, -1, long int>]’
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:195:9:   required from ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:100:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, double*, double*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/fresh/ORB_SLAM2/src/Optimizer.cc:1244:1:   required from here
/usr/local/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/local/include/eigen3/Eigen/src/Core/util/XprHelper.h:816:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
   EIGEN_STATIC_ASSERT((Eigen::internal::has_ReturnType<ScalarBinaryOpTraits<LHS, RHS,BINOP> >::value), \
   ^
/usr/local/include/eigen3/Eigen/src/Core/AssignEvaluator.h:834:3: note: in expansion of macro ‘EIGEN_CHECK_BINARY_COMPATIBILIY’
   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,typename Src::Scalar);
   ^
In file included from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/fresh/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/fresh/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/fresh/ORB_SLAM2/include/Tracking.h:31,
                 from /home/fresh/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/fresh/ORB_SLAM2/include/Viewer.h:25,
                 from /home/fresh/ORB_SLAM2/src/Viewer.cc:21:
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h: In instantiation of ‘class g2o::BaseVertex<3, Eigen::Matrix<double, 3, 1> >’:
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_sba.h:40:35:   required from here
/home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:72: warning: ‘Eigen::AlignedBit’ is deprecated [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
                                                                        ^
In file included from /usr/local/include/eigen3/Eigen/Core:344:0,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/fresh/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/fresh/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/fresh/ORB_SLAM2/include/Tracking.h:31,
                 from /home/fresh/ORB_SLAM2/include/FrameDrawer.h:24,
[... 6 lines cut ...]
In file included from /home/fresh/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35:0,
                 from /home/fresh/ORB_SLAM2/include/LoopClosing.h:34,
[... cut ...]
                 from /home/fresh/ORB_SLAM2/src/LocalMapping.cc:21:
/usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162:37: note: declared here
 EIGEN_DEPRECATED const unsigned int AlignedBit = 0x80;
                                     ^
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

my computer is on ubuntu16.04 
I spent a lot of time but having no idea.i need your help.

a-jahani, 2017-04-09:
Try Eigen 3.2. use "sudo make install" for Eigen

freshn, 2017-04-10:
@a-jahani it works!thanks! But when i try the ORB_SLAM2 by TUM dataset,it runs only 3~4 seconds,followed by a flashback.Following is the result:
// 
fresh@fresh-Inspiron-5548:~/ORB_SLAM2$ ./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml ../下载/rgbd_dataset_freiburg1_desk2/
ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 640

New Map created with 90 points
段错误 (核心已转储)
//

I am sorry that I don't know how to translate the latest Chinese phrase.So I translate it by Google.Maybe it means "Segment error (core dump )".

Thanks for your reply again!

a-jahani, 2017-04-10:
This core dump is mostly because of the eigen. I suggest you :delete all builds and compile pango

freshn, 2017-04-11:
@a-jahani Thanks for your help.I followed your suggestion.But unfortunately when I ran build_ros.sh in orbslam,I got the result as following picture:
[image]
(...Make: *** does not specify the target and can not find the makefile. stop.) 

By the way,when I run build.sh,although there is nothing wrong,the computer would spend at least 20 mins from 59% to the next step.It seems my computer has no response at 59%.I think it is abnormal.
Thanks again.

a-jahani, 2017-04-11:
I think you export the path for ORB_SLAM2 into your ~/.bashrc.
type "gedit ~/.bashrc"
and make sure the address for orbslam2 that you exported there is the same with the one you are compiling.

freshn, 2017-04-11:
@a-jahani Thank you.But I think I have finished what you were suggested to do.
I exported the path by adding this line at the end of .bashrc file:export ROS_PACKAGE_PATH=
${ROS_PACKAGE_PATH}:/home/fresh/ORB_SLAM2/Examples/ROS/ORB_SLAM2 
I also tried calling "rosdep update".But it didn't work.I got the same result when I ran build_ros.sh.
I am at a loss for what to do next.

a-jahani, 2017-04-11:
I cant remember exactly what was the error. You are doing sth out of the instruction.
do the "build. Sh" first then build_ros.sh

freshn, 2017-04-12:
@a-jahani I think I did it in the order you suggested.But it didn't work.Thank you for your help and patience anyway.I would try it again and find more help.
@raulmur Can you  help me?

yingyingqin, 2018-03-10:
@freshn I meet the same problem as you ,have you solved it?段错误 (核心已转储)

ujjwalrathod007, 2018-06-29:
@a-jahani Thank you eigen 3.2 solved my issue? But what is the reason for such error?

freshn, 2018-07-09:
@yingyingqin I 'm sorry... I fixed it that time but I have forgot what methods i tried. It's really been a long time for me. 
But if you follow the tutorial step by step without solving this problem, maybe the memory of your PC is not enough. Try to build it again in another PC first. Hope you can work out it successfully.

chengji253, 2019-05-06:
I met the same problem. And I followed @a-jahani   what he said  . I uninstalled the old eigen and changed  it to eigen3.2. Finally it works. Thanks!