raulmur/ORB_SLAM2 #309: coordinate tranformation

zdczdcc, 2017-04-11:
I know that orbslam's initial coordinate system was random and scale-unknown.But i want to know how to transform the coordinate to another coordinate(from a marker like chessboard)?Is there any paper or any blog?Thanks for your help!

AlejandroSilvestri, 2017-04-21:
@zdczdcc ,
I don't know exactly what you are looking for.  I can refer you to #226 .

There is a recent paper [Metric online monocular SLAM by using a known reference](http://ieeexplore.ieee.org/document/7578496/), not using orb-slam, which uses a chessboard as 6-DOF reference.

zdczdcc, 2017-04-25:
@AlejandroSilvestri ,
Now i'm trying to initialize the orbslam2 with a chessboard(get the camera pose by PnP) to get the definite coordinate system's direction and metric-scale.But when the initialization has done,the graphic optimization would optimize the triangulated mappoints,and it would change the coordinate system defined by the chessboard.Do you have any idea about initialize the orbslam2 with certain coordinate system and scale?

AlejandroSilvestri, 2017-04-25:
I believe you can get the transformation between orb-slam2 coordinate system and your chessboard coordinate system.

Instead of changing orb-slam2's coordinate system, apply such transform to the pose orb-slam2 informs.

zdczdcc, 2017-04-27:
@AlejandroSilvestri 
I don't know the scale of orbslam2's coordinate system,and i only know the scale of the chessboard's coordinate system.In my opinion,we couldn't get the transformation between the 2 coordinate system.If i was wrong,please tell me and give some advice to do that.Thanks a lot!

AlejandroSilvestri, 2017-04-27:
@zdczdcc , you must recognize the chessboard both on image and on orb-slam2 3D map points, in order to get the transformation between orb-slam2 coordinate system and chessboard's.

The paper I cited can enlighten you.

zdczdcc, 2017-04-28:
@AlejandroSilvestri ,
Thank you!I would read the paper you mentioned.