raulmur/ORB_SLAM2 #311: Can I use ORB_SLAM to generate 3D maps?

ColeHoff7, 2017-04-12:
Hello, I have a Bebop3 drone that I want to use ORB_SLAM with so that it can figure out where it is in 3D space. The only problem is I have no idea how to effectively use ORB_SLAM to generate something akin to a 3D map that my drone can use to figure out how far away walls are and things

Any and all advice is appreciated

mslavescu, 2017-04-12:
@ColeHoff7 I'm also looking for something similar here, for SDC LKAS:
https://github.com/OSSDC/OSSDC-LKAS/issues/1

This method is very interesting also:

Voxblox: Building 3D Signed Distance Fields for Planning
https://m.youtube.com/watch?v=PlqT5zNsvwM

m4nh, 2017-04-26:
@ColeHoff7  Take a look here https://github.com/m4nh/skimap_ros.. is our Mapping framework used with another Slam System (SlamDunk). But it is easily pluggable in the ORB_SLAM 2 system.

AlejandroSilvestri, 2017-05-09:
@m4nh , great work Daniele, a very useful and neat integration with orb-slam2.

mslavescu, 2017-05-10:
Thanks @m4nh! 
I'll try it live, once we build the OSSDC DIY smart stereo camera and 360 LIDAR.