raulmur/ORB_SLAM2 #312: How can I use this realtime for my mobile telephone camera?

scp096, 2017-04-13:
Hi,

I'm a newbie at SLAM and Linux system. I followed the tutorial and ran the example TUM Dataset in Ubuntu for a test, it works well. Rightnow I want to upload my camera raw data in realtime to run monocular algorithm. How can I do this? Which interface should I choose?

And how can I make use of my telephone sensor to make a better adjustment?

Thanks a lot for your help.

AlejandroSilvestri, 2017-04-13:
Hi, I don't believe you can use orb-slam2 in realtime in your phone hardware.  Not enough processing power.

Everybody is waiting for a slam method working in their phones.  LSD-SLAM is close enough, working only in tracking mode, slow, 2 or 3 frames per second.

hoverpeter, 2017-04-13:
@AlejandroSilvestri I am able to make it run on iPhone. With some basic optimization, it can run avg. 15fps in full mode.

hoverpeter, 2017-04-13:
I haven't made any effort to port LSD-SLAM to an iPhone. But I really want to. Is there any semi-useful materials out there as guidance?

AlejandroSilvestri, 2017-04-13:
@hoverpeter 
orb-slam2 on iPhone?
That is awesome!
Can you describe your hardware processor?  Only to know the power.
What optimizations did you do?

hoverpeter, 2017-04-14:
It was done on iPhone 7 which is powerful and also on an old iPad Mini, not so powerful but okay. 

Not so much. Started with some basic parallelization.

hoverpeter, 2017-04-14:
@AlejandroSilvestri Could you possibly point me some useful material for porting LSD-SLAM?

scp096, 2017-04-14:
@AlejandroSilvestri @hoverpeter , what about just uploading the camera raw data  to the local PC server when and let the server run it? Maybe I'm just too new to figure out something like a interface to do this.

scp096, 2017-04-14:
@AlejandroSilvestri @hoverpeter , thanks for your guidance. what about just uploading the camera raw data to the local PC server when and let the server run it? Maybe I'm just too new to figure out something like a interface to do this.

AlejandroSilvestri, 2017-04-21:
@scp096 , it is doable.  Actually some students in my class did something like that.  It was a quick and dirty process, I advice against it :)

They took pictures with the phone, with an app that send them to a node.js server, which resend it through some ros plugin to the actual orb-slam2 application in the same server.

It worked, they were looking to relocalize on each image.  Next step will be streaming video.

scp096, 2017-05-06:
@AlejandroSilvestri .Thanks a lot. I took picture and sent every frame to the server and and after some encoding operation I  published  it to the topic. It works for me too. Now I close this issue page.

closed 2017-05-06 by scp096

mslavescu, 2017-05-10:
@AlejandroSilvestri on Android your students could use something like:

https://play.google.com/store/apps/details?id=com.pas.webcam&hl=en

That provides very fast photo or video live streaming over WiFi.

And then use a simple Python OpenCV script to read the live network video stream and publish the frames on ROS network, something like this:

https://github.com/OSSDC/self-driving-car-1/blob/master/docker-instance-scripts/ros_feature_detectio_publisher.py

AlejandroSilvestri, 2017-05-10:
@mslavescu , thank you!

scp096, 2017-05-10:
@mslavescu That's what I did!!

kunyuwang, 2017-08-01:
@hoverpeter Is it possible to share what you did about the basic parallelization to make the orb-slam2 run 15 fps on iPhone 7 or iPad mini.