raulmur/ORB_SLAM2 #316: Get more keyframes from ORBSLAM2

SamLyuubc, 2017-04-17:
Thanks a lot for providing the open source code ORBSLAM2!

        May I ask how to get more keyframes from ORBSLAM2? Currently our dataset is about 2800 frames, but we only get 45 keyframes. Ideally we would like to get more than 1000.

        Thanks a lot!

heimsi, 2017-04-18:
Hello SamLyuubc,
First I would like to point out to you that (according to my knowledge) the GitHub Issue section is not meant for questions like this, but for bug reporting. So I propose you would better ask your question on for example https://stackoverflow.com/ . There you also have a higher chance of getting an answer, as much more people pass by. 
Regarding your question, I think you should read thoroughly the paper of Raul Mur Artál on ORB-SLAM ( you find the link in the readme document) because there it is explained why ORB-SLAM extracts only few keyframes. Actually it extracts a lot of keyframes, as much as possible, but it then in a second step abandons all the keyframes which are redundant ( meaning that they offer only few new information about the scene you are recording) this is actually an important core feature of ORB-SLAM.

SamLyuubc, 2017-04-18:
Thanks for your help Heimsi! I feel very sorry for posting the question at the wrong place. This is my very first time using Github, I should have read through the instruction carefully. Again, Thanks a lot!

a-jahani, 2017-04-20:
You don't need to change it in the code, you can always convert one topic to another using ":=" while running ORBSLAM.
for example:
`rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monocular/demo_cam.yaml /camera/image_raw:=/usb_cam/image_raw`
This converts a published topic in /usb_cam/image_raw to the desired one , /camera/image_raw , in runtime.

AlejandroSilvestri, 2017-04-21:
@SamLyuubc , orb-slam2 create keyframes when they have useful info.  So it restrains itself from adding keyframes when they add no new info to the map.  1 keyframe for 50 frames is normal.

DalderupMaurice, 2017-04-24:
Hi!

Iam having the same issue.. Due to the low keyframes/matches (around 20-50) I always lose track. However, I never see this happen in videos. Any suggestion for this?
Losing track every 2 seconds is not really useful :)

AlejandroSilvestri, 2017-04-24:
@DalderupMaurice ,

People uses to publish videos with orb-slam2 working; when it doesn't work, almost nobody publish the video.

If you're losing track every 2 seconds, something is wrong.  Often, it is due to bad camera calibration, and if you use a wide angle cam, distortion parameters must be correct.  One way to verify the latter is showing the undistorted image, to visually verify if it really is undistorted.

Orb-slam2 do not implement fisheye model, so it doesn't work well with fisheye cams.

Aboute dense mapping: orb-slam2 core uses sparse mapping, with mappoint culling (the fewer points, the better).  Exactly the opposite to dense mapping.

DalderupMaurice, 2017-04-25:
Thank you for your quick response. @AlejandroSilvestri 

I must admit that we used the configuration files from TUM1.yaml, etc on our own Kinect v2 beta. The chance is big that calibration is needed to properly get it working. The problem here is that we have tried several times calibrating and had issues with the IR part (which we cannot skip in the calibration process to fully calibrate the camera), we also do not have a tripod, which leaves us with very variable and incorrect calibration data.
Beyond the calibration, we also have very little knowledge of the further configuration, but we will sure have a look on further configuration.

kitizz, 2017-08-26:
For anyone interested. To answer the original question and expand on the response from @AlejandroSilvestri:
To tweak the number of frames that ORB-SLAM keeps around, there are two critical lines found in `LocalMapping::KeyFrameCulling()`. You're looking for `int nObs` and `if(nRedundantObservations>0.9*nMPs)`. nObs is the number of views a point must appear in to be marked as having redundancies, and changing the `0.9` in the if-statement sets the limit for the fraction of points it sees that have redundancies for it to be removed.

I got the effect I was looking for simply by increasing nObs. If you need more fine-grained control then maybe consider the fraction threshold.