raulmur/ORB_SLAM2 #317: I compile the ORB_SLAM2 error

tianhunzhixin, 2017-04-20:
my gcc version in computer is:
sankuai@w-sankuai-ros:~/Projects/ORB_SLAM2$ gcc --version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4
Copyright (C) 2013 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

And after install the Pangolin, opencv and Eigen3, I run the build.sh. But the error is output, the show is below:
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sankuai/Projects/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sankuai/Projects/ORB_SLAM2/build
[  4%] [  8%] [ 12%] [ 16%] [ 20%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/System.cc.o
Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o
Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o
Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o
Building CXX object CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o
In file included from /home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Tracking.h:31,
                 from /home/sankuai/Projects/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Viewer.h:25,
                 from /home/sankuai/Projects/ORB_SLAM2/src/Viewer.cc:21:
/home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:62:74: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
     typedef Eigen::Map<Matrix<double, D, D>, Matrix<double,D,D>::Flags & AlignedBit ? Aligned : Unaligned >  HessianBlockType;
                                                                          ^
In file included from /home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35:0,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Tracking.h:31,
                 from /home/sankuai/Projects/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Viewer.h:25,
                 from /home/sankuai/Projects/ORB_SLAM2/src/Viewer.cc:21:
/home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_binary_edge.h:59:82: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Di, Dj>, Matrix<double, Di, Dj>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockType;
                                                                                  ^
In file included from /home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:35:0,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Tracking.h:31,
                 from /home/sankuai/Projects/ORB_SLAM2/include/FrameDrawer.h:24,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Viewer.h:25,
                 from /home/sankuai/Projects/ORB_SLAM2/src/Viewer.cc:21:
/home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_binary_edge.h:60:82: warning: ‘Eigen::AlignedBit’ is deprecated (declared at /usr/local/include/eigen3/Eigen/src/Core/util/Constants.h:162) [-Wdeprecated-declarations]
       typedef Eigen::Map<Matrix<double, Dj, Di>, Matrix<double, Dj, Di>::Flags & AlignedBit ? Aligned : Unaligned > HessianBlockTransposedType;
                                                                                  ^
In file included from /home/sankuai/Projects/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34:0,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/sankuai/Projects/ORB_SLAM2/include/LocalMapping.h:26,
                 from /home/sankuai/Projects/ORB_SLAM2/include/Tracking.h:31,
                 from /home/sankuai/Projects/ORB_SLAM2/include/System.h:29,
[... 130 lines cut ...]
/home/sankuai/Projects/ORB_SLAM2/src/LoopClosing.cc: In member function ‘void ORB_SLAM2::LoopClosing::RunGlobalBundleAdjustment(long unsigned int)’:
/home/sankuai/Projects/ORB_SLAM2/src/LoopClosing.cc:670:28: error: ‘usleep’ was not declared in this scope
                 usleep(1000);
                            ^
/home/sankuai/Projects/ORB_SLAM2/src/LocalMapping.cc: In member function ‘void ORB_SLAM2::LocalMapping::Run()’:
/home/sankuai/Projects/ORB_SLAM2/src/LocalMapping.cc:94:28: error: ‘usleep’ was not declared in this scope
                 usleep(3000);
                            ^
/home/sankuai/Projects/ORB_SLAM2/src/LocalMapping.cc:108:20: error: ‘usleep’ was not declared in this scope
         usleep(3000);
                    ^
/home/sankuai/Projects/ORB_SLAM2/src/LocalMapping.cc: In member function ‘void ORB_SLAM2::LocalMapping::RequestReset()’:
/home/sankuai/Projects/ORB_SLAM2/src/LocalMapping.cc:719:20: error: ‘usleep’ was not declared in this scope
         usleep(3000);
                    ^
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Viewer.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2

Anyone can help me???

a-jahani, 2017-04-20:
Easiest way to solve is using eigen 3.2 and make sure delete all build folders and recompile everything from start(pangolin , ... ) . you can add one thing to include as well. you can find your answer in here #144

tianhunzhixin, 2017-04-21:
Thanks for your comments.

closed 2017-04-21 by tianhunzhixin

yohanderose, 2018-12-08:
Hey all, just tried 3.3.2 (2017 Jan) downgraded from 3.4 and its working.
1: in terminal remove your current eigen3 folder with "sudo rm -rf /usr/local/include/eigen3"
2. Download and extract this -> http://bitbucket.org/eigen/eigen/get/3.3.2.tar.bz2
3. cd into the extracted folder and in terminal "mkdir build && cd build" then "cmake .." then "sudo make install"

kmfrick, 2020-05-13:
> Hey all, just tried 3.3.2 (2017 Jan) downgraded from 3.4 and its working.
> 1: in terminal remove your current eigen3 folder with "sudo rm -rf /usr/local/include/eigen3"
> 2. Download and extract this [...]

Please do not do this as it will break packages that depend on your system's Eigen installation. Run ` cmake -DCMAKE_INSTALL_PREFIX=${HOME}/.local/include ..` and include `${HOME}/.local/include` in your `CMakeLists.txt` instead.

nerkn, 2020-08-20:
eigen repo has moved to [gitlab](https://gitlab.com/libeigen/eigen/-/archive/3.3.2/eigen-3.3.2.zip)
`make install ` didnt produced anything
can you be more specific about
` include ${HOME}/.local/include in your CMakeLists.txt`

kmfrick, 2020-08-20:
@nerkn What's the output of `cmake ..` and `make install`?

LongruiDong, 2020-10-15:
On my Ubuntu 18.04 
```
warning: ‘Eigen::AlignedBit’ is deprecated
```
still happen when eigen version down to 3.3.2

aseslamian, 2022-03-10:
> Easiest way to solve is using eigen 3.2 and make sure delete all build folders and recompile everything from start(pangolin , ... ) . anothier option is adding one thing to include as mentioned in here #144 . Please search before opening an issue.

I download earlier version [Eigen3.1.0](https://gitlab.com/libeigen/eigen/-/archive/3.1.0/eigen-3.1.0.tar) but I don't know how to install it from the file?