raulmur/ORB_SLAM2 #318: Viewer Mode Off Issue

shadowver, 2017-04-24:
Hi, 
First of all, thanks for a such awesome software. 
I have tried to run the ORB_SLAM2 on Linux 14.04 with an Intel R200. I notices that there is an option to turn on the default viewer in System.h by setting the bUseViewer = false. However, whenever I turned the viewer off and tried to extract the camera pose
(for instance, cv::Mat pose; 
ORB_SLAM2::System test(string, string, RGBD, false);
pose = test.TrackRGBD(img, depthimg, index);)

the program stop running. It will be stuck on "system resetting" forever unless I forced to quit. 

In addition, I tried use OpenGL to create my own viewer window, the same issue occurs. Any help will be greatly appreciated.

closed 2017-04-25 by shadowver

FPSychotic, 2019-01-18:
please can you help me with r200? No idea how run it.

shadowver, 2019-01-18:
1. get an older version of librealsense (dont know if Intel has already took it off github or not, but 1.12.1 should do)
2. wrap the rgb stream and depth stream with opencv( cv::Mat)
3. input rgb Mat and depth Mat into the system.TrackRGBD. 
following the above three steps should take you there
Cheers~