raulmur/ORB_SLAM2 #319: No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so', needed by '../lib/libORB_SLAM2.so'.  Stop.

howtocodewang, 2017-04-24:
When I tried to compile the project, I always had this problem.
 

> - Generating done
-- Build files have been written to: /home/wangshuo/ORB_SLAM2/ORB_SLAM2/build
make[2]: *** No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so', needed by '../lib/libORB_SLAM2.so'.  Stop.
CMakeFiles/Makefile2:178: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2-

And I also could not find libDBoW2.so in the 

> ../Thirdparty/DBoW2/lib

I also have tried to compile the DBoW2 separately, and then copy the libDBoW2 into the folder. But it didn't work.
So I don't have any idea about that. Can anyone help me? Thanks a lot!

huangjingwei, 2017-04-25:
how did you compile the project?
did you compile in IDE ?

richaagrawal, 2017-05-23:
Did you try running "build.sh"? Is this for ROS or non-ROS version?

howtocodewang, 2017-05-24:
@huangjingwei @richaagrawal Thank you both for your suggestion. I have found out the solution. I didn't compile the project in IDE but run the "build.sh". However, I build this project in Ubuntu 16.04LTS. I tried again to compile the project in Ubuntu 14.04LTS, I made it. So I think this problem may be caused by operation system.

closed 2017-05-24 by howtocodewang

Ichini24, 2020-05-25:
Have the same issue on Ubuntu 18.04

zanazakaryaie, 2020-07-08:
Same issue on Ubuntu 18.04

zanazakaryaie, 2020-07-08:
Solved the issue. It's not related to Ubuntu version. I have OpenCV 4 but the CMakeList of DBoW2 requires OpenCV 3 or 2.4. If none of these versions would be detected, then no lib folder is created to save the library. 
So just remove the version number in find_package(OpenCV), remove CMakeCache file in build directory and re-run build.sh

sdshah09, 2022-07-11:
> Solved the issue. It's not related to Ubuntu version. I have OpenCV 4 but the CMakeList of DBoW2 requires OpenCV 3 or 2.4. If none of these versions would be detected, then no lib folder is created to save the DBoW2 library. So just remove the version number in find_package(OpenCV) ORB_SLAM2/Thirdparty/DBoW2/CMakeLists.txt , then remove CMakeCache file in build directory and re-run build.sh
> 
> There will be some errors in the source codes of Example directory due to some deprecated OpenCV syntaxes. You will need to edit them manually.

make[2]: *** No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so', needed by '../lib/libORB_SLAM3.so'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:711: CMakeFiles/ORB_SLAM3.dir/all] Error 2
make: *** [Makefile:84: all] Error 2 
type of error. I am using Ubuntu LTS 20.04 and OpenCV 4.2 on ORB SLAM 3. I am facing this issue so please let me know if you understand the error. Also I want to know how to generate this .so file

pisces365, 2022-10-10:
Well, I meet the problem as well(make[2]: *** No rule to make target '../Thirdparty/DBoW2/lib/libDBoW2.so', needed by '../lib/libORB_SLAM3.so'. Stop.). And first I try to download `DBoW3` from Github to solve the problem, but it generate a lot of error, mainly focus on orb_salm3 itself can't support `DBoW3`.

So the --Solution-- is cd to ORB_SLAM3/Thirdparty/DBoW2 dir, and cd in to build/, run `cmake ..` and then run `make`.Then you can surprised to find that `libDBoW2.so` is in lib/.Finally, re-run `./build.sh` ,you can finish it 100%.

NO NEED to remove the version number in find_package(OpenCV)     :)

ZefuLin, 2024-07-02:
your method work! thanks!