raulmur/ORB_SLAM2 #322: Callback for Updateted KeyFrames

m4nh, 2017-04-26:
Many mapping frameworks (or other real-time environment renconstruction frameworks) needs to know updated keyframes to rebuild live map. Is there a method to retrieve, during tracking, the updateted subset of keyframes? A callback method is possible to do this asynchronously? Or is just implemented in the library?

Thanks

DalderupMaurice, 2017-05-12:
I'd be interested in this case aswell

laxnpander, 2017-05-15:
Should be easy to implement: Add Callback function to Keyframe class and then trigger it in Optimizer line 767.

DalderupMaurice, 2017-05-16:
Seems like I was looking in the wrong file. Thanks @laxnpander 
So basically it is just the variable "pKF" wich contains the current updated keyframes?
I tried publishing but get a shitload of errors. I have never used c++ before and have trouble understanding the code.

laxnpander, 2017-05-16:
Almost, pKF is a pointer to one of the currently updated keyframes. It will be iterated over all of them. If you have a callback function like "void Keyframe::LocalMappingCallback()" you could access it in the recommended line with "pKF->LocalMappingCallback();" (my tip from memory, currently not able to have a look at the code).
What do you need exactly? Where does updated info go? Ros?

DalderupMaurice, 2017-05-16:
For my thesis my goal is to let a drone fly autonomously. This with the use of a monoculaire camera, ORB-SLAM1 or 2 and ROS. But we are stuck at the last part where we wanted to use ORB (and the ORB camera pose/map points) to be able navigate autonomously. For the moment we litterly have no clue how to achieve this.

laxnpander, 2017-05-16:
Hmmm, gonna be tough if you have no experience in C++. Orbslam is great, but to get the informations you want some handwork is necessary. Best way in my opinion is to pass a callback function as function pointer from your rosnode into the Orbslam. Therefore Orbslam will trigger the function behind the function pointer without thinking about what it will do. The function itself should then be implemented in your rosnode that publishes the Keyframe information directly. I will implement something like this for myself pretty soon, but can't tell you exactly when. 
In addition you will want to get the pointcloud incrementally. As far as I know currently you only get the map as a whole. So you will need to code something there, too.

m4nh, 2017-05-16:
@DalderupMaurice if you want we made an implementation of our Mapping Framework dubbed SKIMAP with ORBSLAM2, the video is this `[video](https://www.youtube.com/watch?v=W3nm2LXmgqE)` and the code is here [SkiMap](https://github.com/m4nh/skimap_ros). Unfortunately due to license problem i can't provide you the integration code but SkiMap is pretty simple to use, we appreciate your feedback about this

m4nh, 2017-05-16:
@DalderupMaurice  to extract 3D points to integrate in the map from a Monocular Slam maybe @laxnpander  can help you better since our integration is based on RGB-D Slam which provide 3D point directly

DalderupMaurice, 2017-05-17:
Apparently SkiMap uses the nonfree version of OpenCV, wich I do not have :P @m4nh 
"fatal error: opencv2/nonfree/features2d.hpp: No such file or directory"
Btw, the video looks great!
And I also have a RGB-D cam in posession on which I can try the project out.

m4nh, 2017-05-17:
@DalderupMaurice no this is a bug! i just removed the unused NonFree include (OpenCV NonFree is used only if you build the SlamDunk module that is our Slam framework, for SkiMap only this is not needed) Try again now with last commit

DalderupMaurice, 2017-05-18:
@m4nh 
Next is: fatal error: skimap_ros/SkimapIntegrationService.h: No such file or directory
 #include <skimap_ros/SkimapIntegrationService.h>

Can we discuss this somewhere else maybe? It is quite irrelevant to ORB SLAM for the moment

DalderupMaurice, 2017-05-19:
@laxnpander 
I have trouble writing the code to publish the keyframes. Whenever I make a callback I do have to initialize a ROS node, but it is never done in the script. Therefor I am unable to create a initialisation of the node because of the missing parameters argc and argv. The only place where ROS is already initialised is in the ros_mono.cc file, but I have no clue how to send the callback data to that class and publish it. My C++ knowledge is way too bad for this. I even had problems with making the callback method and am not sure if it would even work.

laxnpander, 2017-05-30:
@DalderupMaurice Ah sorry, I forgot you. How is it going? Were you able to implement it?

DalderupMaurice, 2017-05-31:
I have managed to figure it out after a couple 100 tries :)

m4nh, 2017-06-02:
@DalderupMaurice hi! can you share your solution for this "issue" please?