raulmur/ORB_SLAM2 #325: Gtk Error When running Stereo Node.

David-Willo, 2017-05-09:
When I run this command
    `rosrun ORB_SLAM2 Stereo Vocabulary/ORBvoc.txt Examples/Stereo/EuRoC.yaml true`
A gtk error occur:
   `ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Stereo

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 435.205
- fy: 435.205
- cx: 367.452
- cy: 252.201
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 20
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1200
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.85272

(Stereo:12419): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported
Trace/breakpoint trap (core dumped)
`
I have no idea of what is happening..Can anyone help?

David-Willo, 2017-05-12:
After removing  gtk+ 2.x, reinstalling Opencv and ROS, everything works well.   Remember to compile the codes using the same gtk version!

closed 2017-05-12 by David-Willo

rosbeginneruser, 2017-08-01:
@David-Willo how did you remove gtk+ 2.x from the project?

David-Willo, 2017-08-01:
@rosbeginneruser Sorry bro, I did not remove it from the project.I am not sure which part is related to gtk2.x and lead to the problem so I completely remove gtk2.x from the system, and reinstall everything. It's not a perfect solution but it works.

jtbon20, 2017-08-01:
@David-Willo Does removing it from the system tank it since there's likely other dependencies? Did it have any adverse effects for you?

David-Willo, 2017-08-01:
It might be a problem if some of the packages in your system depends on gtk2.0, but it's fine in my case. For more detail, check [https://developer.gnome.org/gtk3/stable/gtk-migrating-2-to-3.html](url).