raulmur/ORB_SLAM2 #332: How to find camera settings?

vmhoff, 2017-05-30:
Hi,
I'm running the ORB-SLAM system on my smartphone. I'm using the KITTI00-02 setting file without any modifications and it works well, but quite slow and the system finds too many keyframes. I read in the another issue that wrongs camera settings could do that.
I tried to find my smartphone camera settings with the Matlab's App Camera Calibration, but with these parameters, the system even can't create an initial map.
How could I find the right parameter for my smartphone camera?

Thanks in advance

laxnpander, 2017-05-30:
Can't tell you what tool you might use, but you need the OpenCV calibration standard you can find here:
http://docs.opencv.org/2.4/doc/tutorials/calib3d/camera_calibration/camera_calibration.html
You might use camera_calibration package if you are familiar with ROS. 
http://wiki.ros.org/camera_calibration
If your system creates too many keyframes it might be that your framerate in the settings does not match your real framerate. The fps noted in the settings file has a minor role overall, but it defines some limits that can influence your keyframe creation rate

vmhoff, 2017-05-30:
I'll try to use the OpenCV code for calibration, I'm not familiar with ROS.
I'm using the exactly fps value that the video file informs me. 
Thanks for the answer.

vmhoff, 2017-06-02:
I followed this OpenCV tutorial and I had very similar values with the Matlab ones. I really don't understand why my ORB-SLAM system doesn't work with these parameters.Is someone else running the ORB-SLAM with a smartphone camera?

laxnpander, 2017-06-02:
Hmmm, what Resolution do you use? Smartphone cameras normally have an incredible resolution that no image processing is able to handle. I think you should not get bigger than 1 MPix. 
Also be aware to down scale calibration parameters, if you resize your images.

vmhoff, 2017-06-02:
I'm using 1280x720, which is a bit less than 1MP.
I'll try to change the parameters until I find the optimal parameters that work for me.

adnion, 2017-06-18:
Did it work finally?
I have trouble using my calibration data.
My YAML file looks like this and doesn't work with ORB_SLAM2:

```
image_width: 2448
image_height: 2050
camera_name: prosilica
camera_matrix:
  rows: 3
  cols: 3
  data: [4827.94, 0, 1223.5, 0, 4835.62, 1024.5, 0, 0, 1]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-0.41527, 0.31874, -0.00197, 0.00071, 0]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1, 0, 0, 0, 1, 0, 0, 0, 1]
projection_matrix:
  rows: 3
  cols: 4
  data: [4827.94, 0, 1223.5, 0, 0, 4835.62, 1024.5, 0, 0, 0, 1, 0]
```
At the moment I have to use the example camera settings files. They look totally different. Have a look:
https://github.com/raulmur/ORB_SLAM2/blob/master/Examples/Monocular/EuRoC.yaml

Do you know how to convert the file?

laxnpander, 2017-06-18:
@adnion :
This is a very basic question, you should first check opencv api next time. Your first "data" non zero non one elements are: fx, cx, fy, cy.
Your second "data" is: k1, k2, p1, p2, k3 (irrelevant).
The rest is also irrelevant.

adnion, 2017-06-18:
So you have to rewrite the file on your own after calibration. I didn't get it although I found it in the docs
(http://docs.opencv.org/3.1.0/dc/dbb/tutorial_py_calibration.html).
What I find confusing is:
The camera node starts the package "video_stream_opencv" which uses the above format whereas "rosrun ORB_SLAM2 ... ... ..." uses the other format.
Thanks for your help and the fast response!