raulmur/ORB_SLAM2 #335: Opencv Error when running ROS Examples

Harry-675, 2017-06-03:
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/persistence.cpp, line 740
I met this question,and tried to change the CMakeList.txt,but it didn't work,what should I do?

laxnpander, 2017-06-04:
Do you have different opencv versions installed?

Harry-675, 2017-06-05:
Thanks for attention!
I just installed opencv 3.2.
But I got a warning 
/usr/bin/ld: warning: libopencv_imgproc.so.3.2, needed by ../../../../lib/libORB_SLAM2.so, may conflict with libopencv_imgproc.so.2.4
It seems like that there are also openv 2.4 in my system ,I don't know why.

Harry-675, 2017-06-05:
Besides I can just find opencv3.2 in the /usr/local/lib

laxnpander, 2017-06-05:
I had a similiar problem. The warning sounds like you compiled the ORB SLAM library with a different opencv version than you have now. Check that. And check if you are 100% sure you have no other version installed. Even if you have opencv normally installed and the ros opencv package it may conflict.

MichaelWang1028, 2017-11-27:
I met the same problem, can you tell me how to solve this problem?

OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /home/why/Downloads/opencv-2.4.13/modules/core/src/persistence.cpp, line 740

Harry-675, 2017-11-27:
@MichaelWang1028 Is is happened when you build the ROS example?
If so,I didn't solve the problem in fact,I just didn't use the ROS example for monocular,I wrote a monocular example without ROS,you can find some example on github too.