raulmur/ORB_SLAM2 #336: [rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?

archenroot, 2017-06-05:
I am on Gentoo system using standard ROS within Portage system (not custom overlay, etc.). When trying to execute ./build_ros.sh I get folowing error:
```
zangetsu@ares ~/proj/neural-networks/ORB_SLAM2 $ ./build_ros.sh 
Building ROS nodes
mkdir: adresář „build“ nelze vytvořit: Soubor již existuje
[rosbuild] Building package ORB_SLAM2
Failed to invoke /usr/bin/rospack deps-manifests ORB_SLAM2
[rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?

CMake Error at /usr/share/ros/core/rosbuild/public.cmake:129 (message):
  

  Failed to invoke rospack to get compile flags for package 'ORB_SLAM2'.
  Look above for errors from rospack itself.  Aborting.  Please fix the
  broken dependency!

Call Stack (most recent call first):
  /usr/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
See also "/home/zangetsu/proj/neural-networks/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log".
make: *** Nezadány žádné cíle a žádné makefily nenalezeny. Konec.
```
I reemerged python2_7 multiple times and Gentoo is pulling in Kinetic distribution.

Here are also my config variables available:
```
#ROS
export CMAKE_PREFIX_PATH=/usr
export PYTHONPATH=/usr/lib64/python2.7/site-packages/
export ROS_DISTRO=kinetic
export ROS_ETC_DIR=/usr/etc/ros
export ROS_MASTER_URI=http://ares:11311
export ROS_PACKAGE_PATH=/usr/share/ros_packages
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/zangetsu/proj/neural-networks/ORB_SLAM2/Examples/ROS
export ROS_ROOT=/usr/share/ros
```
I am using folowing version of rosdep:
```
zangetsu@ares ~/proj/neural-networks/ORB_SLAM2 $ rosdep --version
0.11.5
```
Any idea?

archenroot, 2017-06-05:
Additionally I checked following site:

> http://pydoc.net/Python/rosdep/0.11.4/rosdep2.rospack/

which refers to file rospack, so I checked the availability of this file on my system and found following:
```
zangetsu@ares ~/proj/neural-networks/ORB_SLAM2 $ locate rosdep2/rospack.py
/usr/lib64/python2.7/site-packages/rosdep2/rospack.py
/usr/lib64/python2.7/site-packages/rosdep2/rospack.pyc
/usr/lib64/python2.7/site-packages/rosdep2/rospack.pyo
/usr/lib64/python3.4/site-packages/rosdep2/rospack.py
zangetsu@ares ~/proj/neural-networks/ORB_SLAM2 $ diff /usr/lib64/python2.7/site-packages/rosdep2/rospack.py /usr/lib64/python3.4/site-packages/rosdep2/rospack.py
zangetsu@ares ~/proj/neural-networks/ORB_SLAM2 $ 
```

archenroot, 2017-06-14:
This was some issue with the ROS installation as we work now on Gentoo ros-overlay...-> quite experimental

closed 2017-06-14 by archenroot

rohitdavas, 2021-02-16:
I found it to be issue due to anaconda environment.
deactivating conda `conda deactivate` fixed the issue.

N1kYan, 2021-05-21:
`pip install rosdep` to install rosdep into your anaconda env. instead of deactivating it

withnoerr, 2024-04-27:
我的电脑里没有安装anaconda, 也遇到了同样的问题, 我找到了问题原因, 是由于电脑中同时存在多个版本的python, 所以需要将其他版本python删除掉, 重新运行make, 即可解决此问题 
我是从这里得到的灵感[https://github.com/raulmur/ORB_SLAM/issues/109](url)