raulmur/ORB_SLAM2 #337: error: usleep is not declared in this scope

wangzheqie, 2017-06-06:
find that file,and add the following codes to it:

#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>

Jackzhemliu, 2017-06-13:
This is helpful. Thanks.

kuonangzhe, 2017-07-27:
This just solved my problem. Thanks a lot! @wangzheqie

jtbon20, 2017-08-07:
@wangzheqie Why did it work before, but now trying to reinstall I need to add it to nearly all the files?

wangzheqie, 2017-08-10:
so to me, I add it to all files needed without second thinking. And they all work.

gerardsimons, 2017-09-22:
Thanks a lot. Why is this not already in the repo?

paulushcgcj, 2017-11-27:
Why do I need to add this?

I mean I can compile my code on Linux (CentOS 6) perfectly, but why do I need to add this includes when I'm using cygwin, ?

riccitensor, 2017-12-11:
Works, but I had to add it to a bunch of files.

kauevestena, 2017-12-25:
Was necessary to add in 11 files!
But thank you, worked
How does the developers solve this issue?

Aceralon, 2018-03-17:
An easier way is to add `#include <unistd.h>` in `system.h`.

charithforex, 2018-04-30:
@Aceralon Thanks your method just sorted issue really quickly

charithforex, 2018-04-30:
@Aceralon 
Were you able to run this on a camera feed?

ujjwalrathod007, 2018-06-30:
@wangzheqie thanks it solved my error after adding this headers in several files...
But why did I need to do this. ?  What is the reason.?

changshen, 2018-07-18:
I added this for my cygwin project in windows 10 for Code::Blocks
#ifdef __CYGWIN__
#include <sys/unistd.h>
#endif // __CYGWIN__

and the include path of cygwin  c:/cygwin/usr/include should be in search path of compiler.  Where 
c:/cygwin is the install root of cygwin.  And in settings of compilerof cygwin, add __CYGWIN__ in user defined files "#define".  You can do this in project level of in global level.

trehansiddharth, 2018-07-31:
I have replicated that ORB_SLAM2 won't compile on several systems without adding #include <unistd.h> to System.h, as @Aceralon suggested, even after commit #577. Can we incorporate this into the master branch? It seems like it is a simple change?

Alex23013, 2019-11-09:
I had to add it to a bunch of files. But it works . thanks  @wangzheqie

K296, 2020-05-25:
> An easier way is to add `#include <unistd.h>` in `System.h`.

This works but why is this not commited?

> I have replicated that ORB_SLAM2 won't compile on several systems without adding #include <unistd.h> to System.h, as @Aceralon suggested, even after commit #577. Can we incorporate this into the master branch? It seems like it is a simple change?

Why is this suggestion still not incorporated?

Aceralon, 2020-05-25:
> @Aceralon
> Were you able to run this on a camera feed?

Yes

stolpa4, 2020-05-25:
> > I have replicated that ORB_SLAM2 won't compile on several systems without adding #include <unistd.h> to System.h, as @Aceralon suggested, even after commit #577. Can we incorporate this into the master branch? It seems like it is a simple change?
> 
> Why is this suggestion still not incorporated?

Because the repo is dead. Author is lost somewhere

Chatoyant19, 2020-07-16:
> An easier way is to add `#include <unistd.h>` in `System.h`.

it works well! Thank u

balisujohn, 2021-03-21:
> An easier way is to add `#include <unistd.h>` in `System.h`.

This worked for me; thanks for your help!