raulmur/ORB_SLAM2 #340: Reconstruction of the 3D Model of the Environment

AmirCognitive, 2017-06-08:
Thanks for your great package.

I can run it using ROS by camera using mono and RGBD.

However, I am not able to reproduce the last part of the following clip which is about "dense point-cloud reconstructions from estimated RGB-D key-frame poses".

https://www.youtube.com/watch?v=ufvPS5wJAx0&feature=youtu.be&t=2m2s

Could you give me some points on how to reconstruct the 3D model of the environment?

m4nh, 2017-06-08:
@AmirCognitive  sorry but why you are opening another issue again? You can answer/ask in the other opened issues

laxnpander, 2017-06-08:
In short: you can't. At least with the standard. It is from an additional paper proposed by the authors, but not available for public.

AmirCognitive, 2017-06-09:
Okay, so you upload a video on youtube with some nice part added to it in order to attract people, then when people ask how did you do that, you answer is, "In short: you can't". 

Well, in short: you did great!
I am sure your package is very good, but the part that you put something in the video that is not actually exist in your package and according to what you say is impossible, you know, that is not good.

rajputasif, 2017-06-10:
@AmirCognitive I understand  your frustration,  however generating such dense point cloud is not that hard for example use Fast fusion as it can directly be used on camera trajectory

PS. I am not a developer, but I would  like to point out that it is open source project and I can say from my experience  that there comes a time when you are not allowed or restricted to publish some essential piece of code in public.... I hope @laxnpander will agree

laxnpander, 2017-06-10:
@rajputasif @AmirCognitive :
I am not part of the Orb Slam project, just a random guy from the internet. Your frustration is a little bit out of place. Raulmur and his team published what they got, there is also a paper about the dense reconstruction if I remember right. But the code isn't availablt. You can't be angry about that. It might be too experimental for others to use, or millions of other legit reasons.
In addition there are already multiple issues about dense reconstruction you could have used to inform yourself.

laxnpander, 2017-06-10:
Btw. this is raulmurs study about dense reconstruction you saw: http://webdiis.unizar.es/~raulmur/MurTardosRSS15.pdf

shadowver, 2018-01-06:
I have implemented PCL with Orb2 to produce pointcloud. It wont create the dense pointcloud, but it looks much better than the original points which enough for my project. One thing keep in mind that is the current system wont erase the keyframes and mappoints in map() class but only pointers. Hence, the output clouds that with error will also be generated.

rFalque, 2019-11-18:
I know it is very late but I found a script called [generate_registered_pointcloud.py](https://svncvpr.in.tum.de/cvpr-ros-pkg/trunk/rgbd_benchmark/rgbd_benchmark_tools/src/rgbd_benchmark_tools/generate_registered_pointcloud.py) they made which is not documented which does exactly this, you can download it here [https://vision.in.tum.de/data/datasets/rgbd-dataset/tools](https://vision.in.tum.de/data/datasets/rgbd-dataset/tools).

The documentation of the tool itself is not accurate as it is a copy and paste from another file. Also, the script has to be called from the folder of the dataset itself (otherwise the `rbg.txt` and `depth.txt` don't link to the good folder). So for example for me to call it, I did
`~/dev/tests/ORB_SLAM2/Examples/RGB-D/rgbd_dataset_freiburg3_teddy$ python ../generate_registered_pointcloud.py --downsample 5 --nth 10 rgb.txt depth.txt ../../../CameraTrajectory.txt denseMap.ply`

So the dense reconstruction on the teddy dataset produces something like this (here are three files, in color the dense cloud, in green the camera position, and in red the keypoints):
[image]

huonggiangdhtn, 2022-03-09:
[image]
I create point cloud and it seems not good, is there anyone can help me?  I don't know why