raulmur/ORB_SLAM2 #344: Use Lookup table for extracting ORB descriptors

liming0791, 2017-06-12:
Instead of compute 'cos()' and 'sin()' value for rotated ORB pattern, I find that if use a pre-computed lookup table to save different rotations, the speed of extracting ORB descriptors will improve a lot !

Have you considered such a way ? Or finally chose computing rotated pattern for quality or some other things ?

antithing, 2017-06-13:
Hi, I am looking at ways to speed up orbslam2, what kind of increases are you seeing? Would you mind sharing what you have done? Thanks!

liming0791, 2017-06-15:
@antithing ORB_SLAM2 extract ORB descriptors by points array 'bit_pattern_31', this is for degree 0. For other orientations, I computed 8 main orientations and 'hard code' them. Thus avoid calling 'cos()' and 'sin()' when computing descriptors. This reaches 6x ~ 7x speed up for extracting ORB descriptors.

VenkataRamanaS, 2017-08-08:
I am a newbie to computer vision and orb slam trying to improve speed, how do you compute other orientations?
and mainly how is bit_pattern_31 formed initially for degree 0? 
Any resources provided will be highly useful

liming0791, 2017-08-24:
@VenkataRamanaS 

bit_pattern_31 formed as [ x1 y1 x2 y2 ] for one row, represent tow offset around interest point to compute descriptor. 

To compute other orientations, just do 2D rotations and record the new coordinates.

peter-zhao, 2018-05-11:
@liming0791 How much accuracy degeneration or track loss did you observe with the "discrete" version of oBRISK?  If not, I guess it is possible to simply create 18 or 36 lookup tables and use them in ORBExtraction instead :)