raulmur/ORB_SLAM2 #348: FAST corners at eight-scale levels

Toumi0812, 2017-06-13:
Hello everyone,
I know that we extract 1000 corners at eight-scale levels. How are they calculated (for each level, how many)? and to compare keypoints (multiscale) between two frames? 

Thanks

AlejandroSilvestri, 2017-06-19:
Hi @Toumi0812 ,

When using FAST you provide a threshold, but not the quantity of features (or corners) it will find.
So ORB-SLAM2 divide the image in a grid and run FAST on each cell (or subimage).  If it doesn't find enough features, ORB-SLAM2 repeats FAST on that cell with a less strict threshold, hoping to find more features.  There's no third chance.

So ORB-SLAM2 usually extracts more or less 1000 features, but it actually depends on scene and image size.

The procedure repeats on each scale level.

In tracking mode (normal operation mode, with or without mapping)  keypoints are not compared between two frames, but between the actual frame and the map points that are supposed to appear on it.

The scale where ORB-SLAM2 will look for a match is calculated based on distance from actual pose and the map point.

Toumi0812, 2017-06-20:
Hi @AlejandroSilvestri ,
Thank u very much,  but 1000 feature for each level? is there a relation between features extracted in different scales?

AlejandroSilvestri, 2017-06-24:
Hi, I meant a total of 1000 features (not per level), but only as an order of magnitude.

Number of features depends on image.  Usually you'll get less features in higher levels, because images are smaller.

Toumi0812, 2017-06-28:
Hi @AlejandroSilvestri ,
Thanks