raulmur/ORB_SLAM2 #351: Segmentation fault (core dumped)

ahuo002, 2017-06-16:
dicl@ahuo:ORB_SLAM2$ ./Examples/Monocular/mono_kitti Vocabulary/ORBvoc.txt Examp
les/Monocular/KITTI03.yaml /media/dicl/Woko/work_space/KITTI/dataset/sequences/00

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 721.538
- fy: 721.538
- cx: 609.559
- cy: 172.854
- k1: 0
- k2: 0
- p1: 0
- p2: 0
- fps: 10
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 2000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 4541

init done 
opengl support available 
New Map created with 171 points
Segmentation fault (core dumped)

Then, I try the older vision:https://github.com/shomin/ORB_SLAM2, it works(but no map viewer update).

DxsGeorge, 2017-06-16:
Can you try and run it with gdb?

HDaoud, 2017-06-17:
I agree with @DxsGeorge try run it in gdb and check where the error is.
Although I don't think it's related but why are you running the yaml file of sequence 03 with sequence 00? you should use the 00-02 settings file.

ahuo002, 2017-06-20:
@HDaoud @DxsGeorge Thank you. It works now. Maybe I should not change CMakeList.txt with OpenCV3.2.0 and Eigen3.3.4. But the latest verison: "error: 'usleep' was not declared", then add #include<unistd.h> in each file.

InigoCastro, 2018-01-22:
Hello.

I am facing the same problem as you but I did not quite understand from your comments how you managed to fix it. 

In my case I included #include<unistd.h> in every file that prompts the usleep error when building ORB SLAM 2. However, the issue persists and `Segmentation fault (core dumped)` is displayed when I try to run a EuRoC dataset example. I do not know how to make use of gdb as I just learned about this tool.

yingyingqin, 2018-03-09:
I meet the seam problem, but I don't know how to fix it. @ahuo002 @DxsGeorge

richard-elvira, 2018-03-09:
You can run the executable with gdb for view where the core exceptions are. For use it you only need put that follow

gdb Examples/Monocular/mono_euroc

Later the gdb is launched, so you need to run the program with

run here_put_the_parameters_of_orbslam

The code will be executed and when the core dump success you could see in what thread and in what line of code

If you can see the track of the function when it crashes then you can use other command without exit of gdb

bt

And when you finish, you exit to gdb with

quit

yingyingqin, 2018-03-09:
Thank You very much, I will try.

JeffreyYH, 2018-11-28:
The problem is caused by Eigen. Use version 3.2.10 instead of 3.3.X.

Ankita-Singh-21, 2021-03-15:
Thanks @JeffreyYH , it worked for me

ColieShen, 2024-09-06:
Delete all '-march=native' in all cmakelist.txt works for me, including cmakelist in g2o and dbow2 folder.
I am working on ubuntu 22.04, eigen 3.2.10, opencv 3.4.16
`set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3")
# set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native ")
# set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall   -O3 -march=native")`