raulmur/ORB_SLAM2 #355: Required Computer Performances

pilote7107, 2017-06-21:
Hi there, I am using ORB-SLAM2 for my thesis and tuned it for a while now. I must say thanks for it.

I post here about the required performances of computers for running ORBSLAM2. It would be very interesting to make a benchmark per configuration with the output FcomputedPS , and if the founders could advice us about the most important criterions to run faster.
I start (if it has not been done anywhere else)

Configs tested:

Interl(R)Core(TM)i7-4610M CPU @ 3.00GHz
RAM 16.0Go
64bit
Nvidia Quadro K3100M
Windows 7 Pro
Tara Camera 640*480 inputs 30FPS
Ouput Pose 18-20 FPS 
---------------------------------------------------
Intel(R) Core(TM) i7 - 6700HQ CPU @ 2.60 GHz 
RAM 8.00Go
64bit
Windows 10
Nvidia GeForce GTX 1070
Tara Camera 64*480 inputs 30 FPS
Ouput Pose 13-16 FPS

AlejandroSilvestri, 2017-07-08:
@pilote7107 

I believe for a fair comparison it would be better to use the same dataset.

orb-slam2 was designed to maximise performance on general purpose microprocessor.

There are people who worked on improving performance using GPGPU, mainly for ad hoc FAST and BRIEF.

pras2020, 2018-01-03:
@pilote7107 

Did you used GPU with orb slam to achieve this fps.

pras2020, 2018-01-03:
@AlejandroSilvestri 

My first doubt is in the first configuration the RAM is around 16 GB and in second configuration 8 GB. Is it the RAM which is making the difference in output pose.