raulmur/ORB_SLAM2 #356: How to use kitti and euroc dataset to match with CameraTrajectory.txt

pigbreeder, 2017-06-21:
Hi @raulmur 

When I run stero_kitti  and stero_euroc example with the dataset, I find it's difficult for me to use tum benchmark to test the result directly. KITTI and EuRoc provide the ground-truth can't use directly.

Is there any procedure  that convert the result or the ground-truth to use tum benchmark directly?

Thanks.

MrRonSC, 2017-10-06:
Hi @pigbreeder 
Have you figured out how to match the ground truth of KITTI dataset with the result of ORB-SLAM2? I am working on it too but found no detailed instructions.
Thanks.

MichaelGrupp, 2017-10-30:
Feel free to check out my [evaluation software](https://github.com/MichaelGrupp/evo), it is much more flexible than the TUM RGB-D benchmark scripts and supports TUM, KITTI, EuRoC and ROS bags.

For example, you can use the KITTI ground truth directly to compare against FrameTrajectoryKITTIFormat.txt of ORB-SLAM. An example for exactly that use case is in the README.

PYGUAN, 2019-02-06:
Hi @MichaelGrupp 
I use ” evo_rpe kitti 00.txt CameraTrajectory.txt -r trans_part -d 100 -u m" and “evo_rpe kitti 00.txt CameraTrajectory.txt -r angle_deg -d 100 -u m" to evaluate the translation and rotation errors of ORB_SLAM2, however, the results that I get differ greatlly from those in the paper. Why？

MichaelGrupp, 2019-02-06:
@PYGUAN

> results that I get differ greatlly from those in the paper. Why？

I don't know which numbers you mean. But if they are measured like described in the [KITTI challenge](http://www.cvlibs.net/datasets/kitti/eval_odometry.php) I think you need to take all pairs into account, convert to percent (probably `mean error / delta`, aka `mean error / sequence length`) and then average this for different sequence lengths from 100 to 800m.

With `evo_rpe kitti 00.txt CameraTrajectory.txt -r trans_part -d 100 -u m --all_pairs` and `evo_rpe kitti 00.txt CameraTrajectory.txt -r trans_part -d 100 -u m --all_pairs` you should get a mean error of around 1% of 100m. In the KITTI leaderboard, ORB has 1.15%

*But in general:* it's hard to get _exactly_ the same numbers unless you use the same tuning and the same evaluation tool as the author.

PYGUAN, 2019-02-07:
@MichaelGrupp 
Thank you for your reply. Indeed I get a mean error of around 1% of 100m for kitti sequence 00,  but in the paper, the relative translation error trel is 0.7%. Now,I have three questions:
1. Is the translation error of ORB_SLAM2 in the KITTI leaderboard mean error or rmse error?
2. I use "evo_rpe kitti 00.txt CameraTrajectory.txt -r angle_deg -d 100 -u m --all_pairs" to evaluate the rotation error and get a mean error of 0.622963deg/100m for sequence 00. But in the paper it is 0.25 deg/100m.
3.  I use "evo_rpe kitti 00.txt CameraTrajectory.txt -r angle_deg -d 1 -u m --all_pairs" to evaluate the rotation error and get a mean error of 0.083243 deg/m for sequence 00, but in the kitti leaderboard,ORB is 0.0027 deg/m.

MichaelGrupp, 2019-02-07:
> 1. Is the translation error of ORB_SLAM2 in the KITTI leaderboard mean error or rmse error?
This should be documented somewhere.

As I said, you won't get the exact same result unless you generate the same trajectory data as the author and use the same metric. My evo tool can be used to measure errors similar to the KITTI benchmark, but to compare with the leaderboard you should use the official KITTI devkit code.

PYGUAN, 2019-02-07:
@MichaelGrupp 
Okay, thank you!

MichaelGrupp, 2019-02-07:
You're welcome :slightly_smiling_face: