raulmur/ORB_SLAM2 #360: I can not understand the sequence of program.

hyunhoJeon, 2017-06-26:
First, I want to say sorry. Because I have no knowledge of multithreading. Therefore, there is some difficulty in understanding the code. 

I saw the paper and code. And i was started the program on windows but it didn't stop at breakpoint. 
Is this a characteristic of multithreading? Or is it a problem with changing to a Windows version?

And because of this problem, I do not get a clear understanding of how the code works. 
When will the program optimize, mapping, loop closing? 
I saw the paper, but the understanding is not clear. So if you have a description of the sequence of program code-based, please let me know. Thank you :)

TimSC, 2017-07-02:
I've documented the main features in my branch. https://github.com/raulmur/ORB_SLAM2/pull/370

hyunhoJeon, 2017-07-04:
Thanks @TimSC !!
I'm reading your comment. Thank you very much.