raulmur/ORB_SLAM2 #362: Why the error of the stereo is so great?

pssp123, 2017-06-27:
Why the error of the stereo is so great? I have a acurracy calibration data of the stereo.And I test the stereo ORB-SLAM2 in 60㎡ scene,but the error is between 2-30cm.How to improve the acurracy?

pilote7107, 2017-06-29:
Hi!  I am currently using stereo too, most errors come from the calibration. Note that the bigger distorsion you have (ie focal length) the more complicated the rectification is so the input image will be a wrong reflect of the real world => some errors too. 
What camera do you use?

gauxonz, 2017-07-01:
Hi, it sounds that I have the similar situation: #367

paulpaul91, 2017-07-01:
hi,I also find the stereo have a bigger errro,I don't know how to solve

slamhuangxin, 2018-08-06:
@pssp123 @pilote7107 @gauxonz @paulpaul91 hey,guys, do you find the method to solve the problem about the stereo error?

gauxonz, 2018-08-23:
@slamhuangxin Hi my solution now is 
1. First make sure you set a proper `ThDepth` inside the `.yaml` file. This variable mainly controls the farthest depth that you trust on your camera. For example, I'm using the ZED stereo camera with the baseline of 0.12m, then I set `ThDepth` to 80, means I trust all the points from stereo camera with depth smaller than 0.12*80= 9.6m.
2. Initialization matters. Therefore in the [Tracking::StereoInitialization](https://github.com/raulmur/ORB_SLAM2/blob/f2e6f51cdc8d067655d90a78c06261378e07e8f3/src/Tracking.cc#L509) function, you can reject all the point larger than your threshold, by changing
```c++
            if(z>0)
```
to 
```c++
            if(z>0 && z<mThDepth)
```
This works well for me now.

slamhuangxin, 2018-08-23:
@gauxonz 
it's really kind of you.thank you very much.from your reply, i have gotten your solution.it gives me much help,and gets the right result.
best wishes.