raulmur/ORB_SLAM2 #363: Which part is the "PLACE RECOGNITION"?

realitian, 2017-06-27:
[image]

Hi, 
Thanks much for sharing the great SLAM source code.
btw, could anybody answer which part is the "PLACE RECOGNITION" in code?

AlejandroSilvestri, 2017-07-08:
Hi @realitian 

The answer is KeyFrameDatabase.

After object initialization, it is used mainly in  LoopClosing::DetectLoop and Tracking::Relocalization.

realitian, 2017-07-16:
Hi @AlejandroSilvestri 
Thanks for your reply

Btw, I tried to save and load map using shomin's fork and the saving and loading works fine.
But I couldn't work absolutely in monocular mode.
in initialization, it does crash(in addition vertex already exist).
I feel that there missed any processing for monocular current frame when save system.
is there any try to save/load full system?

Cheers

AlejandroSilvestri, 2017-07-17:
@realitian 

I have my own load/save system, working well for monocular only:
[OS1](https://github.com/AlejandroSilvestri/os1)

OS1 is a modified version of orb-slam2.  I believe you should want to grab the serializer code and insert it to your own copy of orb-slam2.

Serializer is mainly in serializer.cpp and serializer.h, but because it uses boost::serializer, it need some code lines appended to the header of some existing classes, like KeyFrame and MapPoint.

realitian, 2017-07-17:
Hi @AlejandroSilvestri 
Thanks a lot of let me know that.
your code worked very well to save and load whole map.
but the tracking had switched into "localization mode" when load existing map by clicking "menu.Guardar mapa" menu.
I tried to switch into 'tracking mode" by clicking checkbox "Tracking, sin mapeo", but I counldn't track scene anymore

AlejandroSilvestri, 2017-07-17:
@realitian 
After loading a map, system goes to relocalization mode.  When orb-slam relocalize itself (when it knows where it is), it switch to tracking mode.  You can unselect "Tracking sin mapeo" to full tracking and mapping.

But beware: the video must first relocalize in order to switch (automatically) to tracking.

realitian, 2017-07-17:
@AlejandroSilvestri 
I recorded my situation as a video, here 
https://drive.google.com/open?id=0B_SVbcquOmxRcC1icjBReTEzZWc

as you can see in the video, when load the map, then camera is frozen.
should it wait for a while?
or the purpose of save/load is only to show the map before recorded?
I wanted to track scene with the recorded map continuously.
e.g. at place 'A', track 'A' and place AR object 'O', and then save all of that.
after that, load 'A'+'O' and track place nearby 'A', then the AR object 'O' will appear automatically without placing.

what was a misunderstanding in my mind?
Cheers

AlejandroSilvestri, 2017-07-17:
@realitian 

I don't know why the camera is frozen after loading the map.  It shouldn't.  I usually use video files, and works fine.  I even load a map and extend the mapping with another video.

This video shows orb-slam on a loaded map, tracking only.
https://www.youtube.com/watch?v=3hW5if_m2Yw

Soon I will update the code, let's see if it works in your case.

realitian, 2017-07-18:
I forgot the press of ' '
perfect!
Thanks again.
:+1:

zgq91, 2017-07-28:
@AlejandroSilvestri Hi,  can you tell me that the route you follow is the route to map or any route on the map.

AlejandroSilvestri, 2017-07-31:
@zgq91 , I'm not sure if I understand your question, what you see on video is tracking on an already mapped zone.

realitian, 2017-08-06:
Hi @AlejandroSilvestri 
How are you?
when I load the exist map (with your code), I couldn't insert a new keyframe that map.
It is impossible to update the loaded map?
Cheers

AlejandroSilvestri, 2017-08-06:
Hi @realitian 

When you load a map, the system goes to tracking only mode.  Change it to tracking and mapping, it will add new keyframes and mappoints.

zgq91, 2017-08-07:
@AlejandroSilvestri When I save the map, it is easy to track lost in the map area. Why?  your video tracking is great.

AlejandroSilvestri, 2017-08-07:
@zgq91 ,

Mapping a whole area can be tricky.  You must map the area from many points of view, from many camera poses.  BRIEF descriptors are used to match features from one pose to other near it.  When you navigate in "tracking only" mode, you can lose tracking when observing the same scene from a different angle.

This has nothing to do with map saving.

zgq91, 2017-08-07:
@AlejandroSilvestri  The saved map contains information such as map point, keyframe . only in places that are close to the keyframe. other map area locations are easy to fail . why is this.  whether it is related to the orb feature, if replaced by surf features will be better.( I do not ask for speed)

AlejandroSilvestri, 2017-08-07:
@zgq91 

I can't tell what is causing to lose tracking.

ORB-SLAM aim to real time.  If no real time is needed, you can use SURF or AKAZE (among others) instead of ORB.  But beware, you should change BoW and use something compatible with your choice, and build your own vocabulary.

Or perhaps use dense monocular slam.

closed 2017-08-08 by realitian

jtbon20, 2017-08-08:
@AlejandroSilvestri What's the best way to change the BoW or make my own? Are there any resources you could point me to? Thanks in advance--