raulmur/ORB_SLAM2 #366: Crash on MacOS serria using rgbd with fr1 dataset.

smallzhan, 2017-06-29:
$./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt TUM1.yaml rgbd_dataset_freiburg1_xyz Examples/RGB-D/associations/fr1_xyz.txt

>ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

>Input sensor was set to: RGB-D

>Loading ORB Vocabulary. This could take a while...
>Vocabulary loaded!

>Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

>ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

>Depth Threshold (Close/Far Points): 3.09294

>-------
Start processing sequence ...
Images in the sequence: 792

>New map created with 833 points
2017-06-29 10:30:06.302 rgbd_tum[64131:17282193] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'nextEventMatchingMask should only be called from the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fffa24dc2cb __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fffb72e748d objc_exception_throw + 48
	2   AppKit                              0x00007fffa06c8e82 -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 4480
	3   libpangolin.dylib                   0x0000000109850fad -[PangolinNSApplication run_step] + 109
	4   libpangolin.dylib                   0x0000000109850986 _ZN8pangolin9OsxWindowC2ERKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEEiib + 294
	5   libpangolin.dylib                   0x0000000109850790 _ZN8pangolin19CreateWindowAndBindENSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEEiiRKNS_6ParamsE + 384
	6   libORB_SLAM2.dylib                  0x00000001076fbc18 _ZN9ORB_SLAM26Viewer3RunEv + 136
	7   libORB_SLAM2.dylib                  0x000000010761c571 _ZNSt3__114__thread_proxyINS_5tupleIJMN9ORB_SLAM26ViewerEFvvEPS3_EEEEEPvS8_ + 113
	8   libsystem_pthread.dylib             0x00007fffb7de593b _pthread_body + 180
	9   libsystem_pthread.dylib             0x00007fffb7de5887 _pthread_body + 0
	10  libsystem_pthread.dylib             0x00007fffb7de508d thread_start + 13
)
libc++abi.dylib: terminating with uncaught exception of type NSException

lieff, 2017-06-29:
It can be libpangolin related crash, not  libORB_SLAM2.

smallzhan, 2017-06-30:
The reason is that Viewer::Run is running in the thread other than main thread. https://github.com/raulmur/ORB_SLAM2/pull/179 solve the problem.

closed 2017-06-30 by smallzhan

zgxsin, 2018-11-20:
Hi, @smallzhan, I have the same problem with you. After pulling from #179, I could not build it. There seems to have some problems with Pangolin. See below please:

`/Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_opencv/modules/highgui/src/cap_ffmpeg_impl.hpp:1679:35: error: use of undeclared
      identifier 'AVFMT_RAWPICTURE'
        if( (oc->oformat->flags & AVFMT_RAWPICTURE) == 0 )
                                  ^
/Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_opencv/modules/highgui/src/cap_ffmpeg_impl.hpp:1913:32: error: use of undeclared
      identifier 'AVFMT_RAWPICTURE'
    if (!(oc->oformat->flags & AVFMT_RAWPICTURE)) {
                               ^
/Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_pangolin/src/video/drivers/ffmpeg.cpp:501:33: error: use of undeclared identifier
      'CODEC_FLAG_GLOBAL_HEADER'
        stream->codec->flags |= CODEC_FLAG_GLOBAL_HEADER;
                                ^
/Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_pangolin/src/video/drivers/ffmpeg.cpp:504:37: warning: 'codec' is deprecated
      [-Wdeprecated-declarations]
    int ret = avcodec_open2(stream->codec, codec, NULL);
                                    ^
/usr/local/include/libavformat/avformat.h:876:5: note: 'codec' has been explicitly marked deprecated here
    attribute_deprecated
    ^
/usr/local/include/libavutil/attributes.h:94:49: note: expanded from macro 'attribute_deprecated'
`#    define attribute_deprecated __attribute__((deprecated))`
            /Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_opencv/modules/highgui/src/cap_ffmpeg_impl.hpp:2207:25: error: use of undeclared
      identifier 'CODEC_FLAG_GLOBAL_HEADER'
            c->flags |= CODEC_FLAG_GLOBAL_HEADER;
                        ^
                                    ^
/Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_opencv/modules/highgui/src/cap_ffmpeg_impl.hpp:2290:9: warning: ignoring return value of
      function declared with 'warn_unused_result' attribute [-Wunused-result]
        avformat_write_header(oc_, NULL);
        ^~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~~
/Users/zhou/MacORB_SLAM2/build/Dependencies/Source/ep_pangolin/src/video/drivers/ffmpeg.cpp:564:39: error: use of undeclared identifier
      'AVFMT_RAWPICTURE'
`

`make[5]: *** [modules/highgui/CMakeFiles/opencv_highgui.dir/src/cap_ffmpeg.cpp.o] Error 1
44 warnings and 16 errors generated.
make[5]: *** [src/CMakeFiles/pangolin.dir/video/drivers/ffmpeg.cpp.o] Error 1
make[4]: *** [src/CMakeFiles/pangolin.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [Dependencies/Stamp/ep_pangolin/ep_pangolin-build] Error 2
make[1]: *** [CMakeFiles/ep_pangolin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[4]: *** [modules/highgui/CMakeFiles/opencv_highgui.dir/all] Error 2
make[3]: *** [all] Error 2
make[2]: *** [Dependencies/Stamp/ep_opencv/ep_opencv-build] Error 2
make[1]: *** [CMakeFiles/ep_opencv.dir/all] Error 2
make: *** [all] Error 2
`

So how did solve it in the end?
Thanks!